Moddable-OpenSource/moddable #224: How two upload example to the Moddable Two

blackhuman, 2019-07-05:
As title says, I can't upload the compile result to the *Moddable Two* board, the esptools.py constantly says "A fatal error occurred: Timed out waiting for packet header".

Before this, I've successfully uploaded some example to the *M5Stack Core* board. *M5Stack Core* board has an embedded CP2012 UART which display the "/dev/cu.SLAB_USBtoUART" port on MacOS. The *Moddable Two* was shipping with the same CP2012 UART program board, but when I plugged it to the *Moddable Two* board and started flash it using command `mcconfig -d -m -p esp32/moddable_two` in the example "network/ble/ios-media-sync" dir, the error "Timed out waiting for packet header" shows up.

I've also tried to push the *flash* button behind the board to try to make the board turn in to the boot mode, but encountered the same error.

There is a question: how to make the board be in the boot mode? and how to get rid of the "Timed out waiting for packet header" error?

bfriedkin, 2019-07-05:
Hello -

Sorry to hear that you are having trouble flashing the app. Could you please confirm that the programming board is plugged into the Moddable Two the correct way? The pin labels on the programming board should match the labels on the Moddable Two. See the attached picture below:

[image]

Please let us know. We can diagnose further if the board is properly connected.

Regards,
Brian

blackhuman, 2019-07-06:
Still didn't work. My *Moddable Two* was shipping with an pre-programmed test program which can drag and drop the small component. (Show as following picture)

[image]

And I plugged the programming board as follows: (TX->RX,RX-TX)

[image]

To diminish the uncertain factor, I do use the origin esp32 *hello world* example to test, with Serial Port and Serial baud rate set.

[image]

Unfortunately, the result is same as before, with same error. (Note. the blue LED blinks when starting to flash)

￼￼￼[image]

blackhuman, 2019-07-06:
Shocking! when I record the video, I found that what the printing on the board is "moddable one", which means that I got an esp8266 based *Moddable One* instead of the esp32 based *Moddable Two*. Maybe the vendor made a mistake to shipping the wrong board? I'll confirm with vendor for this.

bfriedkin, 2019-07-08:
Following up privately to resolve shipment issue.

- Brian

closed 2019-07-08 by bfriedkin

davidmoshal, 2019-07-26:
I have the same problem, and I do have the Moddable Two.
#236