Moddable-OpenSource/moddable #225: Moddable Two: GPIO Pin 25?

zgramana, 2019-07-06:
The status of pin 25--a very important pin for sound--on the moddable two is confusing, and would appreciate clarification.

The [product info page for the Two](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/moddable-two.md#pinout) clearly shows pin 25 routed out to a pad labelled both `GP25` and `SPK +`. Weirdly, however, while the photo _looks_ like my Two in most respects, the product photo's PCB silkscreen actually says `moddable one`.

[image]

On the Moddable Two that I received, pin 25 is just not broken out at all, full stop.

[image]

Did I get a wrong revision of the board, or the right one? If the later, why was the pin 25 breakout removed? I can understand why you might have decided to not break it out as an I2C internal DAC speaker driver circuit, but not breaking it out at all just makes it very hard to use I2C (which the ESP's excel at). While technically the matrix allows I2C from other pins, I haven't been about to get it to work and several knowledgable ESP32 sources have said it's unlikely to work in most scenarios.

Everything about the Two, an integrated ESP32 + cap touch screen, screams to be paired with I2S sound and an SD card. Though other the moddable sdk doesn't allow it for the Two, I've been able to get an sdcard working via HSPI by using the Arduino IDE instead (why no HSPI support for the Two in the Moddable SDK?), but the lack of pin 25 access has left me crestfallen.

Is there any reason to _not_ solder a wire onto the pin 25 pad directly (other than soldering skill ability)?

zgramana, 2019-07-07:
I can get mono using the I2S device (0) + the internal DAC using pin 26 (which also worked without modification with the `piu/sound` demo). However, I'm not able to get anything out of an external [MAX98357A DAC/amp](https://www.sparkfun.com/products/14809?_ga=2.160987961.585427172.1562435070-159683055.1561939606). I can see the signal from the Two on the breakout board's pins, using pins 26, 22, and (pin 25 alternate) 16, but I'm not getting audio out. Further troubleshooting will require protocol analysis with the scope to see if there are timing issues.

waldopt, 2019-07-10:
The image of the Moddable Two in the Getting Started guide was an internal version that included testing for audio output. The initial release of the Moddable Two did not include that. We anticipate adding GPIO 25 to the external header in a feature rev of the Moddable Two.

zgramana, 2019-07-14:
I hope you'll update that photo so that others don't get confused. That image is the bulk of the extent documentation on the hardware side of things, given there are no published schematics, etc., so I'm sure others like me give it a lot of attention.

closed 2019-07-14 by zgramana

lprader, 2019-07-15:
@zgramana, I've updated the photo in the pinout to show the board we actually released. It'll show up in our next open source update.