Moddable-OpenSource/moddable #226: VIN for Moddable Two

zgramana, 2019-07-07:
I'm trying to power my Two off of battery instead of USB. I initially assumed `VIN` would be 3.3v, and that the USB 5v was just regulating down to 3.3v. I plugged a beefy 3.3v pack into VIN and got nothing. Is VIN 5v? Is VIN just giving us 5v from USB with a safety diode? Or am I just missing something?

zgramana, 2019-07-07:
From some of the [other docs](https://rntlab.com/question/can-i-power-esp32-with-5v-via-the-vin-pin/) around, it seems that in the [reference design](https://dl.espressif.com/dl/schematics/ESP32-Core-Board-V2_sch.pdf) the `VIN` header needs to be fed 5v since the regulator ties directly to the 3.3 pin on ESP32 module itself.

waldopt, 2019-07-10:
On Moddable One and Moddable Two VIN supports powering the board with a 5V source. VIN goes to the voltage regulator and is stepped down to 3.3v to power the ESP32. VIN only supports 5v.

zgramana, 2019-07-14:
@waldopt I think I'm not alone in planning to power my Moddable Two project with a 3.3v LiPo cell. Unfortunately, I either have to add a boost converter to get as close I can get to 5.0v, or I have to manually by-pass the regulator and jumper directly to the VIN pin of the ESP32 module. Neither option is attractive.

For the next revision of the board, I hope you will consider opting for a more user-friendly LDO regulator like the [AP2112k-3.3](https://www.digikey.com/en/datasheets/diodes-incorporated/diodes-incorporated-ap2112). This would give us the ability to source 3.3-5.0v (and with a little wiggle room).

closed 2019-07-14 by zgramana