Moddable-OpenSource/moddable #227: Add HSPI support for the Moddable Two via the SDK

zgramana, 2019-07-07:
Presumably VSPI is used by the TFT and touch controller, but HSPI (available on GPIO pins `32`, `14`, `13`, `27`) also works great for adding microSD storage of images, sounds, logs, etc. It's usable when programming the Two via Arduino, but would much rather stay in the Moddable SDK and use it there! 😉

phoddie, 2019-07-08:
You may use any pins for SPI you like with the Moddable SDK on ESP32. The Moddable Two manifest defines the SPI pins to use  [here](https://github.com/Moddable-OpenSource/moddable/blob/1a8514cea3915180d19330f9fbe9660c9f9d8b7a/build/devices/esp32/targets/moddable_two/manifest.json#L15):

		"spi": {
			"miso_pin": 12,
			"mosi_pin": 13,
			"sck_pin": 14,
		},

You can change that to use other pins in your application manifest by overriding the SPI pins configuration, for example:

		"spi": {
			"mosi_pin": 23,
			"miso_pin": 19,
			"sck_pin": 18
		},

closed 2019-07-08 by phoddie

phoddie, 2019-07-08:
The information above is to configure the SPI pins. The SPI port (HSPI or VSPI) is passed as an argument to `modSPIConfig` when establishing a new SPI connection.

zgramana, 2019-07-08:
Thanks, that's a big help!