Moddable-OpenSource/moddable #229: 'this.disconnect() no function!' error

adarkazanli96, 2019-07-13:
Hi, I have a class that extends BLEClient. I have it set to call this.disconnect() after receiving a total of 10 notifications from the onCharacteristicNotification() callback. However, whenever this.disconnect() gets called, I receive: "# Break: HeartRateClient.prototype.?: call disconnect: no function!' indicating that this.disconnect() is not a function. To try to debug this, I looked at the examples included in the sdk, and the only one that uses this.disconnect is the advertiser example which I can't really run since I can't even connect it to anything. My app does however disconnect automatically if i turn off the bluetooth server it's being connected to. But I would still like the ability to disconnect it myself by calling the method, please help!

bfriedkin, 2019-07-13:
@adarkazanli96 -

Thank you for reaching out.

You can disconnect the peripheral by calling the `close()` method on the `device` passed to the `onConnected()` callback:

```
const HR_SERVICE_UUID = uuid`180D`;
const HRM_CHARACTERISTIC_UUID = uuid`2A37`;

let count = 0;

class HeartRateClient extends BLEClient {
	onConnected(device) {
		this.device = device;
		device.discoverPrimaryService(HR_SERVICE_UUID);
	}
	onCharacteristicNotification(characteristic, value) {
		if (++count == 10) {
			count = 0;
			this.device.close();
		}
	}
}
```

FYI - There is also a BLEClient `close()` function. Calling BLEClient.prototype.close() disconnects all peripherals **and** closes the client instance.

Regards,
Brian

closed 2019-07-13 by bfriedkin

adarkazanli96, 2019-07-13:
@bfriedkin worked like a charm! thank you for the fast response!