Moddable-OpenSource/moddable #234: Windows: xsl.EXE exits with error code 0xD on self-build and program build

nodebotanist, 2019-07-21:
Commit: c92dff58 of `public` branch

Error message: 
```
### 'C:\Users\Nodebotanist\Projects\moddable\build\tmp\win\release\tools\modules\commodetto\Bitmap.xsb': invalid file
NMAKE :  U1077: 'C:\Users\Nodebotanist\Projects\moddable\build\bin\win\release\xsl.EXE' : return code '0xd'
Stop.

C:\Users\Nodebotanist\Projects\moddable\build\makefiles\win>call C:\Users\Nodebotanist\Projects\moddable\build\bin\win\release\mcconfig -m -p x-win C:\Users\Nodebotanist\Projects\moddable\tools\xsbug\manifest.json
```

This error message shows repeatedly when building.

This prevents me from building the XS application suite and programs for the XS engine

bfriedkin, 2019-07-21:
@nodebotanist -

Apologies for the inonvenience.

Our last GitHub push of the Moddable SDK included many new features. There were significant updates to both the tools and runtime. Please try rebuilding your Moddable tools and apps clean by deleting your `%MODDABLE%\build\bin` and `%MODDABLE%\build\tmp` directories and then rebuild the tools:

```
cd %MODDABLE%\build\makefiles\win
build
```

After the tools build completes, you can rebuild and run your Moddable apps.

Regards,
Brian

nodebotanist, 2019-07-23:
hmmmm... after deleting those directories, windows gives me the following on rebuild:

C:\Users\Nodebotanist\Projects\moddable\build\makefiles\win>call C:\Users\Nodebotanist\Projects\moddable\build\bin\win\release\mcconfig -m -p x-win C:\Users\Nodebotanist\Projects\moddable\tools\xsbug\manifest.json
'C:\Users\Nodebotanist\Projects\moddable\build\bin\win\release\mcconfig' is not recognized as an internal or external command,
operable program or batch file.

nodebotanist, 2019-07-23:
Ah, caught it-- didn't actually delete the directories in Windows. Thanks!

closed 2019-07-23 by nodebotanist