Moddable-OpenSource/moddable #236: A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

davidmoshal, 2019-07-26:
Hi, just got my shiny new Moddable Two.  
Attached programmer.    
Followed install directions.  
Checked that I can see /dev/ttyUSB0
Checked that I am part of the dialout group.
Ran this command:
```
cd ${MODDABLE}/examples/piu/balls
mcconfig -d -m -p esp32/moddable_two
```
Results in: 
```
CC /home/davem/dm/moddable/build/tmp/esp32/moddable_two/debug/idf/main/main.o
AR /home/davem/dm/moddable/build/tmp/esp32/moddable_two/debug/idf/main/libmain.a
LD /home/davem/dm/moddable/build/tmp/esp32/moddable_two/debug/idf/xs_esp32.elf
esptool.py v2.6
Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x10000 )...
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
/home/davem/esp32/esp-idf/components/esptool_py/Makefile.projbuild:62: recipe for target 'flash' failed
make[1]: *** [flash] Error 2
```

davidmoshal, 2019-07-26:
not sure if this helps, but:
```
screen /dev/ttyUSB0 115200
```
resulted in:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:656
ho 0 tail 12 room 4
load:0x40078000,len:6192
load:0x40080000,len:5076
entry 0x40080264
```
whereas:
```
screen /dev/ttyUSB0 921600
```
was not human readable.

I suspect that connecting at the lower speed might work.
How best to configure the sdk to do that?

phoddie, 2019-07-27:
To try flashing at a lower speed, edit  [sdkconfig.defaults](https://github.com/Moddable-OpenSource/moddable/blob/public/build/devices/esp32/xsProj/sdkconfig.defaults). There are three lines to change:

```
CONFIG_ESPTOOLPY_BAUD_115200B=y
CONFIG_ESPTOOLPY_BAUD_921600B=
CONFIG_ESPTOOLPY_BAUD=115200
```

The debugger also runs at 921600 by default. To change that edit [make.esp32.mk](https://github.com/Moddable-OpenSource/moddable/blob/public/tools/mcconfig/make.esp32.mk):

```
DEBUGGER_SPEED ?= 115200
```

In our experience, the main cause of USB failures at 921600 is a bad USB cable. If it does work for you at 115200 you may want to try another cable.

phoddie, 2020-10-20:
This issue has been dormant for over a year. Closing now. If the issue is still active, please reopen with addition details. Thank you.

closed 2020-10-20 by phoddie