Moddable-OpenSource/moddable #238: Windows build incorrectly specifies (requires?) /Zp1

vvuk, 2019-07-28:
The makefiles for the Windows build specify `/Zp1`, to request 1-byte packing (basically no alignment).  This is not compatible with any Windows SDKs, as the headers there require default packing rules.  `xsPlatform.h` includes `winsock2.h`, which requires the Windows SDK headers.

This error looks like this from the build:
```
C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um\winnt.h(2487): error C2118: negative subscript
```
because it's a C assert macro in winnt.h which fails in that way (the actual error is shown in the header in a comment).

I'm not sure whether 1-byte packing is required to build the tools, but I don't see the equivalent gcc option in the linux makefile rules (`pack-struct`), or even any packed attributes.  Removing `/Zp1` from all makefiles makes the build succeed, not sure yet if it works.

bfriedkin, 2019-07-29:
@vvuk -

Thank you for the report. I have been able to confirm that the tools builds succeed and apps run from my Windows 10 build host when removing the `/Zp1` alignment directives from the make files. I am following the Windows setup and build instructions from our [Getting Started](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md#windows) document. In particular I am building from the the "x86 Native Tools Command Prompt for VS 2017" command line console. I can build with and without the alignment directive.

Given that you aren't able to build with the `/Zp1` alignment directive, could you please share your build configuration? We should better understand what the differences are.

Regards,
Brian

bfriedkin, 2019-08-06:
@vvuk -

The `/Zp1` alignment directives have been removed from the Windows make files.

Regards,
Brian

closed 2019-08-06 by bfriedkin