Moddable-OpenSource/moddable #239: !!! xs: failed to allocate 1024 bytes for slots !!!

wilberforce, 2019-07-30:
I'm unclear under `creation:` in the manifest on what setting to increase.

[image]

infact for this esp32 project - I'm note clear where the current settings are getting inherited from?

The slots are curently 18432.

in the clock example I see:

https://github.com/Moddable-OpenSource/moddable/blob/public/contributed/modClock/manifest.json#L13-L16

and here:
https://github.com/Moddable-OpenSource/moddable/issues/193#issuecomment-493091095

however I don't understand how this relates to `slots`

Thanks!

wilberforce, 2019-07-30:
Re-reading it looks like I need to increase static  from 32k.

Will try tomorrow.

phoddie, 2019-07-30:
The "creation" in the manifest is the total available to slots, chunks, and the stack. The stack is static but the division between slots and chunks is somewhat dynamic. Each slot is 16 bytes. A stack frame entry is a slot.

The ESP32 has tons of memory (at least relative to the ESP8266). You can easily bump to 64 KB or more (hard limit around 128 KB due to address banks on ESP32 hardware architecture). We keep the default small to keep ourselves working under a constraints to avoid getting sloppy.

wilberforce, 2019-07-31:
@phoddie 
Thanks. Do you have some sort of _rule of thumb_ that you apply for what a resonable amount of head space for % usage for slots, chunks, stack?

closed 2019-08-11 by wilberforce

phoddie, 2019-08-13:
There's no simple answer because projects are so different. That said...

If you are running for a while through all the usual operations of the project and the total of (slots peak) + (chunks peak) + (stack size) is within 2 KB of your static allocation, it is kind of tight. Neither the chunk heap nor the slot heap can fragment, so you never lose space to that.

Also, if the garbage collector is running often (dozens of times a second) that suggests allocating more memory for the virtual machine could improve performance -- though, again, it really depends on the project characteristics.

wilberforce, 2019-08-13:
@phoddie Thanks