Moddable-OpenSource/moddable #242: Required version of python?

danfinlay, 2019-08-08:
The [getting started guide](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md#mac) does not specify a python version.

Today mac ships with Python 2.7 as its default, but it's reaching end of life, and I was unable to connect to my ESP32, so I was thinking maybe I need to use a specific version of Python?

bfriedkin, 2019-08-08:
@danfinlay -

Sorry to hear about the ESP32 connection issue. The ESP-IDF Getting Started Guide, [macOS Toolchain Setup](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/macos-setup.html) states that "ESP-IDF will use the version of Python installed by default on macOS."  The statement is vague but implies that whatever Python version is installed by default should be compatible.

FWIW - I am using Python 2.7.15 and haven't experienced any connection issues:

```
Brians-MacBook-Air:moddable brianfriedkin$ python --version
Python 2.7.15
```

Can you provide any more details on the connection failure?

Regards,
Brian

danfinlay, 2019-08-10:
That rules out the python theory. I’ll share more soon.

closed 2019-08-10 by danfinlay