Moddable-OpenSource/moddable #243: Bad copy of `ESP8266_RTOS_SDK`?

danfinlay, 2019-08-08:
I've got a few moddable devices, been hopping between them (happy to use whichever one sets up the easiest on my mac ;) ).

Right now I'm on the `moddable-zero`, and I'm trying to build with the command `mcconfig -d -m -p esp/8285` but I get this error:

```
make: *** No rule to make target `/Users/danfinlay/esp/ESP8266_RTOS_SDK/lib/libcirom.a', needed by `/Users/danfinlay/Documents/Development/moddable/build/tmp/esp/debug/lib/lib_a-setjmp.o'.  Stop.
```

As far as I can tell, this implies the file `moddable/build/tmp/esp/debug/lib/lib_a-setjmp.o` is looking for this `ESP8266_RTOS_SDK/lib/libcirom.a` file, which does not exist in the `v3.0-rc1` tag that [the setup instructions suggest](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md#esp8266-setup).

Did that tag get updated? Am I missing an install step?

bfriedkin, 2019-08-08:
@danfinlay -

That is unexpected. If you are building for the Moddable One platform, please make sure to specify that platform on the `mcconfig` command line:

```
cd $MODDABLE/examples/piu/balls
mcconfig -d -m -p esp/moddable_one
```

Also please try a clean build. Delete the `$MODDABLE/build/tmp/esp` and `$MODDABLE/build/bin/esp` directories before running `mcconfig`. If that doesn't work, please share your complete build output so that we can compare to a working build.

Regards,
Brian

danfinlay, 2019-08-23:
Sorry for the long wait - I am doing this as a hobby for now :)

Deleting those build did appear to help at first, until I realized this just meant it re-built other files, before hitting the same error.

When I connect the Moddable One and run the commands you suggest, I get this output:

```
# cpp abi.cpp
# cc cont.S
# cc cont_util.c
# cpp core_esp8266_main.cpp
# cc core_esp8266_noniso.c
# cc core_esp8266_phy.c
# cc core_esp8266_postmortem.c
# cc core_esp8266_si2c.c
# cc core_esp8266_timer.c
# cc core_esp8266_wiring.c
# cc core_esp8266_wiring_digital.c
# cc core_esp8266_wiring_pwm.c
# cpp Esp.cpp
# cc heap.c
# cc libc_replacements.c
# cc spiffs_cache.c
# cc spiffs_check.c
# cc spiffs_gc.c
# cc spiffs_hydrogen.c
# cc spiffs_nucleus.c
# cc time.c
# cc umm_malloc.c
# cpp Schedule.cpp
# cc bsearch.c
# cc e_fmod.c
# cc rtctime.c
# cc tinyprintf.c
make: *** No rule to make target `/Users/danfinlay/esp/ESP8266_RTOS_SDK/lib/libcirom.a', needed by `/Users/danfinlay/Documents/Development/moddable/build/tmp/esp/debug/lib/lib_a-setjmp.o'.  Stop.
make: *** Waiting for unfinished jobs....
# cc tinyuart.c
```

bfriedkin, 2019-08-25:
@danfinlay -

Thank you for following up. I took a closer look at the error message and why the build is looking for the `libcirom.a` file in a directory that doesn't exist.

Our ESP8266 [make file](https://github.com/Moddable-OpenSource/moddable/blob/public/tools/mcconfig/make.esp.mk) includes the following:

```
ifneq ("$(wildcard $(ESPRESSIF_SDK_ROOT)/components/esp8266/lib/libcirom.a)", "")
	LIBCIROM_PATH = $(ESPRESSIF_SDK_ROOT)/components/esp8266/lib/libcirom.a
else
	LIBCIROM_PATH = $(ESPRESSIF_SDK_ROOT)/lib/libcirom.a
endif
```

One explanation for the error could be that your build is not finding the `$(ESPRESSIF_SDK_ROOT)/components/esp8266/lib/libcirom.a` file and dropping in the `else` case above, which sets the `LIBCIROM_PATH` to the path displayed in the make error message.

I found that the `libcirom.a` file does not exist at the expected location if I just clone the ESP8266_RTOS_SDK repo, but not checkout the `v3.0-rc1` version. If I clone the repo and checkout the `v3.0-rc1` version, the `libcirom.a` file does exist in the expected directory.

Please double check that you checkout the `v3.0-rc1` version after cloning the ESP8266_RTOS_SDK repo and navigating into the `ESP8266_RTOS_SDK` directory:

```
cd ~/esp
git clone https://github.com/espressif/ESP8266_RTOS_SDK.git
cd ESP8266_RTOS_SDK
git checkout v3.0-rc1
```

Regards,
Brian

danfinlay, 2019-08-26:
Thanks for the follow-up!

I swear I had checked out that tag, but it gave me commit `a2b75212` previously. Now I'm on commit `fd63b5e5`. Still no `lib` folder, but maybe the `if` block will catch next time I get to try (hopefully soon).

Feel free to close this for now, I can reopen when I get my next try opportunity.

bfriedkin, 2019-08-26:
The ESP8266 build seems to work for us. I also confirmed that the ESP8266_RTOS_SDK / v3.0-rc1 checkout hasn't changed in some time.

I'll go ahead and close this case for now. Fingers crossed you'll be up and running soon enough with the correct SDK version. If that isn't the case, please let us know.

Regards,
Brian

closed 2019-08-26 by bfriedkin