Moddable-OpenSource/moddable #244: #error "WiFi configuration check: WARNING, WIFI_RX_BA_WIN should not be larger than WIFI_DYNAMIC_RX_BUFFER_NUM!"

aakoch, 2019-08-10:
I didn't get very far in setting up my Moddable 2. I can't even run the hello world example. I get this:
```
CC /Users/aakoch/projects/moddable/build/tmp/esp32/moddable_two/debug/idf/esp32/wifi_init.o
CC /Users/aakoch/projects/moddable/build/tmp/esp32/moddable_two/debug/idf/main/main.o
/Users/aakoch/esp32/esp-idf/components/esp32/wifi_init.c:24:2: error: #error "WiFi configuration check: WARNING, WIFI_RX_BA_WIN should not be larger than WIFI_DYNAMIC_RX_BUFFER_NUM!"
 #error "WiFi configuration check: WARNING, WIFI_RX_BA_WIN should not be larger than WIFI_DYNAMIC_RX_BUFFER_NUM!"
  ^
make[2]: *** [wifi_init.o] Error 1
make[1]: *** [component-esp32-build] Error 2
make[1]: *** Waiting for unfinished jobs....
AR /Users/aakoch/projects/moddable/build/tmp/esp32/moddable_two/debug/idf/main/libmain.a
cp: /Users/aakoch/projects/moddable/build/tmp/esp32/moddable_two/debug/idf/xs_esp32.map: No such file or directory
cp: /Users/aakoch/projects/moddable/build/tmp/esp32/moddable_two/debug/idf/xs_esp32.bin: No such file or directory
```

Here's my session:
```
Sat 10 11:23:02 esp32 > cd esp-idf                       
Sat 10 11:23:06 esp-idf > gs                        
On branch release/v3.1
Your branch is up to date with 'origin/release/v3.1'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)

	modified:   components/coap/libcoap (untracked content)

Untracked files:
  (use "git add <file>..." to include in what will be committed)

	components/asio/
	components/bootloader/subproject/components/
	components/bt/controller/
	components/bt/host/
	components/esp_wifi/
	components/expat/expat/
	components/lwip/lwip/
	components/mqtt/
	components/protobuf-c/
	components/unity/
	examples/build_system/

no changes added to commit (use "git add" and/or "git commit -a")
Sat 10 11:23:07 esp-idf > echo $IDF_PATH          
/Users/aakoch/esp32/esp-idf
Sat 10 11:23:24 esp-idf > pwd             
/Users/aakoch/esp32/esp-idf
Sat 10 11:23:28 esp-idf > python -m pip install --user -r $IDF_PATH/requirements.txt   
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Requirement already satisfied: setuptools in /Users/aakoch/Library/Python/2.7/lib/python/site-packages (from -r /Users/aakoch/esp32/esp-idf/requirements.txt (line 4)) (32.3.1)
Requirement already satisfied: pyserial>=3.0 in /Users/aakoch/Library/Python/2.7/lib/python/site-packages (from -r /Users/aakoch/esp32/esp-idf/requirements.txt (line 8)) (3.4)
Requirement already satisfied: future>=0.15.2 in /Users/aakoch/Library/Python/2.7/lib/python/site-packages (from -r /Users/aakoch/esp32/esp-idf/requirements.txt (line 9)) (0.17.1)
Requirement already satisfied: cryptography>=2.1.4 in /Users/aakoch/Library/Python/2.7/lib/python/site-packages (from -r /Users/aakoch/esp32/esp-idf/requirements.txt (line 10)) (2.7)
Requirement already satisfied: asn1crypto>=0.21.0 in /Users/aakoch/Library/Python/2.7/lib/python/site-packages (from cryptography>=2.1.4->-r /Users/aakoch/esp32/esp-idf/requirements.txt (line 10)) (0.24.0)
Requirement already satisfied: enum34; python_version < "3" in /Users/aakoch/Library/Python/2.7/lib/python/site-packages (from cryptography>=2.1.4->-r /Users/aakoch/esp32/esp-idf/requirements.txt (line 10)) (1.1.6)
Requirement already satisfied: ipaddress; python_version < "3" in /Users/aakoch/Library/Python/2.7/lib/python/site-packages (from cryptography>=2.1.4->-r /Users/aakoch/esp32/esp-idf/requirements.txt (line 10)) (1.0.22)
Requirement already satisfied: cffi!=1.11.3,>=1.8 in /Users/aakoch/Library/Python/2.7/lib/python/site-packages (from cryptography>=2.1.4->-r /Users/aakoch/esp32/esp-idf/requirements.txt (line 10)) (1.12.3)
Requirement already satisfied: six>=1.4.1 in /Users/aakoch/Library/Python/2.7/lib/python/site-packages (from cryptography>=2.1.4->-r /Users/aakoch/esp32/esp-idf/requirements.txt (line 10)) (1.10.0)
Requirement already satisfied: pycparser in /Users/aakoch/Library/Python/2.7/lib/python/site-packages (from cffi!=1.11.3,>=1.8->cryptography>=2.1.4->-r /Users/aakoch/esp32/esp-idf/requirements.txt (line 10)) (2.19)
Sat 10 11:23:46 esp-idf > echo $PATH               
/Users/aakoch/projects/moddable/build/bin/mac/release:/Users/aakoch/Library/Python/3.7/bin:/Users/aakoch/esp32/xtensa-esp32-elf/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/MacGPG2/bin:/usr/local/share/dotnet:/opt/X11/bin:~/.dotnet/tools:/Library/Frameworks/Mono.framework/Versions/Current/Commands:/usr/local/bin/
Sat 10 11:24:03 esp-idf > ls /dev/cu.*          
/dev/cu.Bluetooth-Incoming-Port /dev/cu.SLAB_USBtoUART
Sat 10 11:24:23 esp-idf > echo $UPLOAD_PORT       
/dev/cu.SLAB_USBtoUART
Sat 10 11:24:32 esp-idf > cd ${MODDABLE}/examples/helloworld  
Sat 10 11:24:49 helloworld > mcconfig -d -m -p esp32           
CC /Users/aakoch/projects/moddable/build/tmp/esp32/debug/idf/esp32/wifi_init.o
CC /Users/aakoch/projects/moddable/build/tmp/esp32/debug/idf/main/main.o
/Users/aakoch/esp32/esp-idf/components/esp32/wifi_init.c:24:2: error: #error "WiFi configuration check: WARNING, WIFI_RX_BA_WIN should not be larger than WIFI_DYNAMIC_RX_BUFFER_NUM!"
 #error "WiFi configuration check: WARNING, WIFI_RX_BA_WIN should not be larger than WIFI_DYNAMIC_RX_BUFFER_NUM!"
  ^
make[2]: *** [wifi_init.o] Error 1
make[1]: *** [component-esp32-build] Error 2
make[1]: *** Waiting for unfinished jobs....
AR /Users/aakoch/projects/moddable/build/tmp/esp32/debug/idf/main/libmain.a
cp: /Users/aakoch/projects/moddable/build/tmp/esp32/debug/idf/xs_esp32.map: No such file or directory
cp: /Users/aakoch/projects/moddable/build/tmp/esp32/debug/idf/xs_esp32.bin: No such file or directory
Sat 10 11:24:54 helloworld >
```

What other troubleshooting steps should I take?

aakoch, 2019-08-10:
Targeting my mac with `mcconfig -d -m -p mac` works fine.

aakoch, 2019-08-11:
There was a commit 17 days ago that added 2 checks. 
https://github.com/espressif/esp-idf/commit/6af07b800a70da78cb0b147099451cfec2c2c822#diff-a708f3e1ab8cbe468fe732dee9ba09c5

I commented the checks out and am able to compile and run now, but what are they for?

bfriedkin, 2019-08-11:
@aakoch -

Sorry for the trouble. We've been able to reproduce the issue. Indeed it seems that Espressif recently updated the v3.1 release, including the change you note above.

We are working through the issue. Thank you for your patience.

Regards,
Brian

bfriedkin, 2019-08-13:
@aakoch -

We released a Moddable SDK update that resolves this issue. The SDK now requires the more recent ESP-IDF v3.2.2 stable release. You'll need to remove the ESP-IDF v3.1 directory and replace that with a clone the ESP-IDF v3.2.2 branch:

```
cd ~/esp32
git clone -b v3.2.2 --recursive https://github.com/espressif/esp-idf.git
```

Then update your Moddable SDK and rebuild your Moddable ESP32 `helloworld` app clean by first deleting the `$MODDABLE/build/tmp/esp32` and `$MODDABLE/build/bin/esp32` directories before running `mcconfig`. Note that because you're building against a newer version of the ESP-IDF, the toolchain may prompt you to enter configuration options. If this happens, accept the defaults.

The [ESP32](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md#esp32-mac) section in the [Getting Started](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md) document has been updated for v3.2.2.

Regards,
Brian

aakoch, 2019-08-17:
Thanks. And while I was able to run hello world, the mini_drag project I was working on stopped working. 

I had to make some adjustments to the main.js to get it to be able to drag, but now when I run it I get: `/Users/aakoch/projects/moddable/examples/commodetto/mini-drag/main.js (26) # Break: /Users/aakoch/projects/moddable/examples/commodetto/mini-drag/main.js: get screen: undefined variable!
unhandled exception`

The `screen` variable is not longer defined.

aakoch, 2019-08-17:
commodetto/text used to work but it too is failing. So probably all of the commodetto examples are broken.

aakoch, 2019-08-17:
Maybe because of the change to line 168 in modules/piu/MC/piuMC.js?:
let screen = global.screen ? global.screen : require("screen");

bfriedkin, 2019-08-17:
@aakoch -

We released a Moddable SDK update yesterday. It looks like you might be running an older version. I am able to build and run the text and progress examples against this version of the SDK on my Moddable Two:

```
mcconfig -d -m -p esp32/moddable_two
```

Please try updating and let us know if that resolves the screen issue. Also, could you please share the changes you needed to make to the mini_drag example app?

Regards,
Brian

aakoch, 2019-08-17:
I did a pull, deleted 2 build directories, re-ran make (twice) and I'm still getting the error. I'll probably try nuking the esp and moddable directories next.

I forked the repo in anticipation of creating a pull request for my changes to the mini-drag app, but wanted to make sure my changes worked with your latest changes first.

aakoch, 2019-08-18:
It helps if I use the correct command. 🙄 

`mcconfig -d -m -p esp32/moddable_two`

NOT 

`mcconfig -d -m -p esp32`

closed 2019-08-18 by aakoch