Moddable-OpenSource/moddable #246: Allow setting a timeout when setting up an I2C connection

nodebotanist, 2019-08-17:
Allow the user to pass in a `timeout` parameter that will account for clock stretching, slow reads, etc from I2C devices.

phoddie, 2019-09-06:
@nodebotanist - In #237 you noted that you were going to try implementing a solution here. Any update?

nodebotanist, 2019-09-18:
Sorry, had to take a break from everything for a bit. Started work back up again.

andycarle, 2021-02-09:
This was implemented for ESP8266 and ESP32 back in 8e39234b7cf047b0475d1e9a74548551ced142b4. Since it has been implemented and this issue has been idle for some time, I'm going to go ahead and close this out.

Please feel free to re-open this or start up a new issue if there are any problems with the timeout implementation!

&nbsp;- Andy

closed 2021-02-09 by andycarle