Moddable-OpenSource/moddable #248: Documentation suggestion for Getting Started Guide

mikew67, 2019-08-18:
In the Moddable SDK – Getting Started guide, In the ESP32 setup sections (for all hosts?), at step 4 you say:

> Download and untar the ESP32 GCC toolchain. Copy the extracted xtensa-esp32-elf directory into your ~/esp32 directory.

There are **two** xtensa-esp32-elf directories, at the top level of the expanded tar and the first level down.  Please add a note to make sure the user copies the top level.

Stumbled on this by accident and was _interesting_ to unravel...

Mike

lprader, 2019-08-19:
I've added a note to the getting started guide. It'll show up in our next open source release.

phoddie, 2019-08-19:
Documentation update is live

closed 2019-08-19 by phoddie