Moddable-OpenSource/moddable #250: Flashing examples on esp32/moddable_two

tomjohnson3, 2019-08-19:
Reading the docs and related github issues, I'm left with the impression that it *should* be possible to build/flash examples to the moddable_two without having to press the "flash" button on the board.

However, on Windows 10, if I don't press the flash button during the "Connect ...___..." phase, I will get a timeout error:

`A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
make: *** [/home/thomas/esp/esp-idf/components/esptool_py/Makefile.projbuild:63: flash] Error 2
`

(Pressing the flash button during the Connect phase *does* flash and run/debug the app correctly.)

Is this the expected behavior? If so, it might help to update the "getting started" docs to reflect this necessary step.

phoddie, 2019-08-19:
You are correct that you should be able to flash the a Moddable Two device without pressing the flash button. There is circuitry on the board to allow for that.

We have seen rare situations where that does not work and it is necessary to press the flash button. There's no single reason that we're aware of for that. You might try a different USB cable (a shorter one, if you are using a particularly long one). If nothing works, we can try swapping out your board in case there is a problem there (we do test every board before shipping it out by flashing the demo app it arrives with, so it worked for us).

tomjohnson3, 2019-08-22:
okay - here's a datapoint: i tried a shorter cable on the same USB port that i was using before...and still no luck auto-connecting.

however, i moved to a different USB port and it worked.

thinking about it more (without opening up my desktop), i think the first (more convenient) port is not directly connected to the motherboard - while the second one is (or is much closer).

anyway - it's happily auto-connecting now.

closed 2019-08-22 by tomjohnson3

phoddie, 2019-08-23:
Congratulations!

It is mysterious that it failed on one port but not the other. We should add that to our trouble shooting guide, as it might help others.