Moddable-OpenSource/moddable #251: struggling with relative imports

dckc, 2019-08-21:
I have the following layout:

```
src/kernel/index.js
test/test-kernel-xs/main.js
test/test-kernel-xs/manifest.json
```

In `main.js` I'd like to use `import buildKernel from '../../src/kernel/index';` because that way my editor (vs-code) will know what I'm talking about.

I can't figure out what to put in `manifest.json` to get `mcconfig` to play along (with cwd = test/test-kernel-xs). Clues?

With manifest 1 below, I can build and run, but I get:
 
```
# Break: (host): module "../../src/kernel/index" not found
```

with manifest 2, `mcconfig -d -m -p lin` says `### Error: No such file or directory`.
strace output includes:
```
[pid 27019] mkdir("/home/connolly/projects/moddable/build/tmp/lin/debug/mc/test-kernel-xs", 0755) = -1 EEXIST (File exists)
[pid 27019] mkdir("/home/connolly/projects/moddable/build/tmp/lin/debug/mc/test-kernel-xs/modules", 0755) = -1 EEXIST (File exists)
[pid 27019] mkdir("/home/connolly/projects/moddable/build/tmp/lin/debug/mc/test-kernel-xs/modules/../../src/kernel", 0755) = -1 ENOENT (No such file or directory)

```

### Manfest 1

```json
{
    "include": [
        "$(MODDABLE)/examples/manifest_base.json"
    ],
    "modules": {
        "*": [
            "./main",
            "../../src/kernel/*"
        ],
        "@agoric/harden": "../../node_modules/@agoric/harden/dist/harden.esm",
        "@agoric/make-hardener": "../../node_modules/@agoric/make-hardener/index",
        "@agoric/nat": "../../node_modules/@agoric/nat/dist/nat.esm",
        "tape-promise/tape": "../../node_modules/tape-promise/index.compiled",
        "onetime": "../../node_modules/onetime/index"
    }
}
```

### Manifest 2

```json
{
    "include": [
        "$(MODDABLE)/examples/manifest_base.json"
    ],
    "modules": {
        "*": [
            "./main"
        ],
        "../../src/kernel/*": "../../src/kernel/*",
        "@agoric/harden": "../../node_modules/@agoric/harden/dist/harden.esm",
        "@agoric/make-hardener": "../../node_modules/@agoric/make-hardener/index",
        "@agoric/nat": "../../node_modules/@agoric/nat/dist/nat.esm",
        "tape-promise/tape": "../../node_modules/tape-promise/index.compiled",
        "onetime": "../../node_modules/onetime/index"
    }
}
```

I guess I can attach a .zip of my working dir...

dckc, 2019-08-21:
[SwingSet.zip](https://github.com/Moddable-OpenSource/moddable/files/3523508/SwingSet.zip)

phoddie, 2019-08-22:
> I'd like to use import buildKernel from '../../src/kernel/index';

The module specifier in the Moddable SDK is not a path. Trying to make it behave as one isn't going to be easy.

Environment variables are supported in the manifest (the use of `$(MODDABLE)` is an example of that). Perhaps define one for your src/kernel directory (e.g. `$SK`), then your editor can resolve that (`import foo from "$SK/foo"`).

closed 2019-08-22 by phoddie

dckc, 2019-08-22:
vs-code doesn't seem to grok environment variables... but using [jsconfig.json](https://code.visualstudio.com/docs/languages/jsconfig) looks feasible.

dckc, 2019-08-22:
I can do stuff like `$SK` in the xs-specific parts of what I'm doing, but the goal I'm working on is to get a bunch of existing code, originally developed using node.js, to work in xs.

Maybe there's a way to transpile the import paths... the inverse of [babel-plugin-module-resolver](https://www.npmjs.com/package/babel-plugin-module-resolver)... maybe [babel-plugin-import-redirect](https://www.npmjs.com/package/babel-plugin-import-redirect).

Ouch.

phoddie, 2019-08-22:
> ...but the goal I'm working on is to get a bunch of existing code, originally developed using node.js, to work in xs.

Good luck. Moving code written for a large environment with a countless runtime options and a huge tooling ecosystem to a small environment designed embedded devices is a perilous undertaking. Going the other way is generally easier.

Consequently, I would start by finding a away to make the code work well for the more constrained target (e.g. XS). Once that works well, then go back to figure out how to create a single code base that works well for both environments.

dckc, 2019-08-25:
It was more the linter than the editor that was confused about relative imports. I found a way to make it happy. It involves changes such as the following in `kernel/kernel.js`, but my collaborators seem OK with it:

```diff
-import { makeLiveSlots } from './liveSlots';
+import { makeLiveSlots } from 'kernel/liveSlots';
```

https://github.com/dckc/SwingSet/commit/bcc368e12e04ec8503c2c34a18e1e7f3b6c20295

dckc, 2019-09-21:
The [platform docs](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/XS%20Platforms.md) show that `fxFindModule` is platform-specific. The default implementation seems to handle `./` and `../` but stops there, without support for `../../`:

https://github.com/Moddable-OpenSource/moddable/blob/public/xs/sources/xsPlatforms.c#L177-L187