Moddable-OpenSource/moddable #252: commonjs example? xsc -m gives error: too many files

dckc, 2019-08-21:
I can't get commonjs support to work. Help?!

I added this to my manifest:

```
    "commonjs": [
        "defined",
        "through"
    ],
```

but `xsc` doesn't seem to like `-m`:

```
# xsc through.xsb
/home/connolly/projects/moddable/build/bin/lin/release/xsc /home/connolly/projects/agoric/SwingSet/node_modules/through/index.js -m -d -c -e -o /home/connolly/projects/moddable/build/tmp/lin/debug/mc/test-kernel-xs/modules -r through
# error: too many files# error: too many files!
```

phoddie, 2019-08-22:
CommonJS modules are not supported. They are non-standard. XS supports standard ECMAScript modules only.

The `-m` option to `xsc` was removed some time ago. The documentation apparently was not updated. We'll update that. Apologies for the misdirection.

closed 2019-09-06 by phoddie

dckc, 2019-09-06:
I see related documentation tweaks in c19ed09 and 8d0e42c, and I'm not sure this is worth much bother, but one tiny nit remains: `-m` is still in [xsc arguments][1]:

```
xsc file [-c] [-d] [-e] [-m] [-o directory] [-p] [-r name] [-t directory]
```

[1]: https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/tools/tools.md#arguments-3

phoddie, 2019-09-06:
The `-m` will be gone in the next update. Thanks for your attention to detail.