Moddable-OpenSource/moddable #257: debugging xs with VS Code: v8 inspector proxy

dckc, 2019-08-31:
I'm interested to give this a try... not urgently, but I'm interested in any advice...

[VS Code debugging][vs1] seems to use v8's inspector protocol.
xsbug has a network protocol; [doCommand][] provides a reasonably concise summary.

Building a proxy to act like v8 inspector and translate to/from xsbug's network protocol seems like a manageable amount of work, if we're only worried about the 80 to 95% of the protocol that's enough to get by and not every corner case.

[vs1]: https://code.visualstudio.com/docs/nodejs/nodejs-debugging
[doCommand]: https://github.com/Moddable-OpenSource/moddable/blob/public/tools/xsbug/DebugBehavior.c#L1129

cc @dtribble

dckc, 2019-11-06:
meanwhile, I figured out vs code mac config to build, debug SwingSet on xs.

https://github.com/dckc/moddable/tree/vs-code-debug-c
b31d03273111c927f418ebc251e33b3ea0894321

dckc, 2020-02-22:
I found `sudo tcpflow -i any -C -g port 5002` a particularly useful tool to watch the debugger and the vm talk to each other.

source: https://superuser.com/a/23187 Jan 20 '17

closed 2023-06-27 by phoddie

dckc, 2023-06-28:
nifty!

A pointer to how to use it would be great. Or just to the relevant commit. I'm struggling to find it.

phoddie, 2023-06-28:
Apologies. I didn't intend to mark this as Completed. I closed it as it had gone (very) cold.