Moddable-OpenSource/moddable #258: Symbol.keyFor() as in @agoric/marshal gives undefined

dckc, 2019-08-31:
When I run this with `mcconfig -d -m -p lin`, I get `same? false actual type: undefined`:

```js
export default function main() {
  var canonical = Symbol.for('s');

  const actual = Symbol.keyFor(canonical);
  const same = actual === 's';
  trace(`same? ${same} actual type: ${typeof actual}\n`);
}
```

It's very odd, because it's based on the [built-ins/Symbol/keyFor/arg-symbol-registry-hit.js][1] test262 test, which passes:

```
~/projects/test262/test$ ~/projects/moddable/build/bin/lin/debug/xst built-ins/Symbol/keyFor/arg-symbol-registry-hit.js 
### built-ins/Symbol/keyFor/arg-symbol-registry-hit.js (sloppy): OK
### built-ins/Symbol/keyFor/arg-symbol-registry-hit.js (strict): OK
# 0:00:00
100% 2/2
```
[1]: https://github.com/tc39/test262/blob/master/test/built-ins/Symbol/keyFor/arg-symbol-registry-hit.js

context: I'm struggling with the `Symbol.for('sym1')` test case in test-marshal.js https://github.com/Agoric/SwingSet/blob/master/test/test-marshal.js#L33-L36

phoddie, 2019-08-31:
Good find. This only happens for symbols created from strings stored internally as `XS_STRING_X_KIND` -- which is, more or less, strings that are known at build time.

The fix is straightforward. In xsSymbol.c change lines 153 & 154 as follows:

```
	if (key && ((key->kind == XS_KEY_KIND) || (key->kind == XS_KEY_X_KIND)) && ((key->flag & XS_DONT_ENUM_FLAG) == 0)) {
		mxResult->kind = (key->kind == XS_KEY_KIND) ? XS_STRING_KIND : XS_STRING_X_KIND;
```

FWIW, a quick review of the other functions in xsSymbol.c suggests that they handle this case correctly..

We'll push the fix to the repository in the coming days.

phoddie, 2019-09-02:
The fix has been pushed.

dckc, 2019-09-02:
confirmed: fixed in 002cbb7c370

closed 2019-09-02 by dckc