Moddable-OpenSource/moddable #259: cannot coerce undefined to object! in template literal

dckc, 2019-08-31:
The template literal expression here throws; is that by design? I'm looking thru various specs and docs and stuff and I can't quite tell whether this OK. The diagnostic is misleading, since neither `Object(undefined)` nor `String(undefined)` throws.

```js
export default function main() {
  const undef = undefined;
  const o = Object(undefined);
  const s0 = String(undefined);
  const s = `undef:${undef}`;
  trace(s + '\n');
}
```

If I try to make my own template tag, the 2nd arg seems to get `undefined` rather than an array of values. The exception comes on the `parts.push` line, presumably when evaluating `args[i - 1]`:

```js
export default function main() {
  const undef = undefined;
  const o = Object(undefined);
  const s0 = String(undefined);
  const s = allstr`undef:${undef}`;
  trace(s + '\n');
}

function allstr(quasis, args) {
  const parts = [quasis[0]];
  for (let i = 1; i < quasis.length; i += 1) {
    parts.push(String(args[i - 1]), quasis[i]);
  }
  return parts.join('');
}
```

context:
https://github.com/Agoric/SwingSet/blob/a74f608035bbb33f7cc6846c69fd16993131bc48/src/kernel/liveSlots.js#L150

phoddie, 2019-09-01:
Thank you for the report.

XS does behave differently than the engines in browsers for this case. The exception you see os generated when trying to call `toString` on `undefined`. The exception is correct when calling `toString` on `undefined`, but it appears that `toString` should not be invoked in that case.

It looks like the issue hinges on the notes in [12.2.9.6](https://tc39.es/ecma262/#sec-template-literals-runtime-semantics-evaluation):

> The string conversion semantics applied to the Expression value are like `String.prototype.concat` rather than the `+` operator.

erights, 2019-09-02:
The important part of that is it bottoms out in the behavior of the internal ToString function
https://tc39.es/ecma262/#sec-tostring

phoddie, 2019-09-02:
> The important part of that is it bottoms out in the behavior of the internal ToString function

@erights - yes, exactly. I didn't have it quite right above. Patrick noted that XS took a short cut on this step, by trying to call the objects's `toString` which is not quite `ToString`.

> 6. Let last be ? ToString(sub).

@dckc - A fix is committed. The change adds a new opcode making it necessary to rebuild tools as well as projects.

dckc, 2019-09-02:
I can confirm that the `undef:${undef}` problem is fixed in 54361c6bd777 .

The `allstr` template tag symptoms persist, but that's not blocking my work; IOU a separate issue.

closed 2019-09-02 by dckc