Moddable-OpenSource/moddable #260: redefinition of 'NSOffState'

KarimIbrahim, 2019-08-31:
Hi, I'm getting many redefinition errors when I "**make**" moddable tools in **moddable/build/makefiles/mac**. This is the error I'm getting:
`# cc parser.c.o
In file included from /Users/karimibr/Projects/personal/moddable/modules/piu/PC/mac/piuField.m:21:
In file included from /Users/karimibr/Projects/personal/moddable/modules/piu/PC/mac/piuPC.h:24:
In file included from /System/Library/Frameworks/Cocoa.framework/Headers/Cocoa.h:13:
In file included from /System/Library/Frameworks/AppKit.framework/Headers/AppKit.h:15:
In file included from /System/Library/Frameworks/AppKit.framework/Headers/NSActionCell.h:8:
/System/Library/Frameworks/AppKit.framework/Headers/NSCell.h:80:34: error: redefinition of 'NSOffState'
static const NSControlStateValue NSOffState /*NS_DEPRECATED_WITH_REPLACEMENT_MAC("NSControlStateValueOff", 10_0, 10_13)*/ = NSControlStateValueOff;
                                 ^
/System/Library/Frameworks/AppKit.framework/Headers/NSCell.h:75:34: note: previous definition is here
static const NSControlStateValue NSControlStateValueOff = 0;
                                 ^
/Users/karimibr/Projects/personal/moddable/modules/piu/PC/mac/piuPlatform.h:38:32: note: expanded from macro 'NSControlStateValueOff'
#define NSControlStateValueOff NSOffState`

I'm using:
macOS High Sierra 10.13.6
XCode 10.1 (10B61)

KarimIbrahim, 2019-09-01:
I resolved it by commenting out these lines in piuPlatform.h:
`
36 //#ifndef MAC_OS_X_VERSION_10_14
 37 //#define NSControlStateValueOn NSOnState
 38 //#define NSControlStateValueOff NSOffState
 39 //#endif
`
NSOnState and NSOffState where already defined in the AppKit framework headers.

bfriedkin, 2019-09-03:
@KarimIbrahim -

Thank you for reporting this issue. I'm glad you were able to find a way to workaround the problem.

So that we can hopefully address this issue for other developers, we have a small change to `piuPlatform.h` we'd like you to confirm, if convenient for you:

Simply change `MAC_OS_X_VERSION_10_14` in the original file to `MAC_OS_X_VERSION_10_13`:

```
#ifndef MAC_OS_X_VERSION_10_13
#define NSControlStateValueOn NSOnState
#define NSControlStateValueOff NSOffState
#endif
```

This change seems to work as expected on macOS 10.12.6 and 10.14.5. It would be great to know if the change works for you on macOS 10.13.6. You can confirm the change by rebuilding your Moddable tools.

Thanks,
Brian

KarimIbrahim, 2019-09-03:
Hey Brian,

I reverted my change and applied your suggested fix and it worked. Please note that my command line tools version is 10.1.0.0.1.1539992718. It might be affecting the headers I have on my mac.

bfriedkin, 2019-09-04:
@KarimIbrahim -

Thank you for the quick confirmation!

Indeed there are multiple variables in play. We use the availability macros here to help support older macOS SDK versions, which has worked reasonably well.

Regards,
Brian

closed 2019-09-04 by bfriedkin