Moddable-OpenSource/moddable #261: eventual send operator syntax

dckc, 2019-09-07:
If I wanted to add support for this emerging syntax, do you have any advice on how to go about it?

for reference...
https://github.com/Agoric/proposal-infix-bang
https://github.com/Agoric/transform-bang
https://github.com/Agoric/acorn-infix-bang
https://github.com/tc39/proposals/blob/master/stage-0-proposals.md
https://github.com/tc39/tc39-notes/blob/master/meetings/2019-07/july-25.md#infix-bang

phoddie, 2019-09-07:
There's no "how to" guide for adding new syntax to XS. ;)

If I understand the proposal, the proposal is syntax that expands to operations already supported in JavaScript. If that is correct, the most straightforward approach might be to implement it as a preprocessor. With that working, it should be straightforward to modify `mcconfig` to invoke it before compiling JavaScript files.

dckc, 2019-09-08:
To clarify: the goal includes supporting this syntax in runtime eval.

closed 2022-10-20 by phoddie