Moddable-OpenSource/moddable #262: Build errors

zgramana, 2019-09-14:
Using latest `public` on macOS [10.14.6], I've started to get build errors in my project:
```text
/path/to/moddable/build/tmp/mac/debug/mc/project/mc.xs.c:503:4: error: use of undeclared identifier 'fx_require'
        { fx_require, 1, -32698 },
          ^
/path/to/moddable/build/tmp/mac/debug/mc/project/mc.xs.c:509:4: error: use of undeclared identifier
      'fx_require_weak'
        { fx_require_weak, 1, -32360 },
          ^
# mcrez resources
/path/to/moddable/build/tmp/mac/debug/mc/project/mc.xs.c:4504:121: error: use of undeclared identifier
      'fx_require'
  ...{.ID = -32698, .flag = 0x8f, .kind = XS_CALLBACK_X_KIND}, .value = { .callback = { fx_require, NULL } }},
                                                                                        ^
3 errors generated.
```
I updated `esp-idf` to `release/v3.3` hoping that would resolve it, but no luck.

Any suggestions?

zgramana, 2019-09-14:
Re-building the tools after the `git pull` was the missing step:

```
cd ${MODDABLE}/build/makefiles/mac
make
```
Recommend adding an "Upgrade" section to the moddable "Getting Started" docs.

closed 2019-09-14 by zgramana

phoddie, 2019-09-14:
Glad you solved your problem. 

> Recommend adding an "Upgrade" section to the moddable "Getting Started" docs.

Agreed. We can improve this.