Moddable-OpenSource/moddable #263: AudioOut stop() & enqueue AudioOut.Flush both lockup Sim and device

zgramana, 2019-09-14:
Trying to cancel enqueued samples, and calling either `stop` or `enqueue(stream, AudioOut.Flush)` results in a hung device (moddable 2) or simulator (macOS).

phoddie, 2019-09-14:
@zgramana, if you would provide an example, that would ensure we are looking into the same issue and help us to isolate the problem. Thanks.

zgramana, 2019-09-16:
@phoddie I've sent your PR #265, which both reproduces the problem and provides a needed example for the use of `AudioOut` (which is woefully under-documented given is rather impressive capabilities--especially as a mixer and for sound+event/effect synchronization).

It appears there are a couple of issues which this example can reproduce. I will file separate issues for them in case they are not the same root cause.

zgramana, 2019-09-16:
### Steps to reproduce this issue

1. Launch a debug session in simulator
1. Establish expected behavior:
    1. Press play
    1. Wait for melody to finish playing
    1. Observe the following trace log message: `=> Playing sound...done`
1. Observe the defect
    1. Press play
    1. Immediately press play again
    1. Observe the following trace log message: `=> Playing sound=> Stopping sound`
    1. Notice that no further interaction with the simulator is possible
    1. Notice that the debugger is unresponsive to the 'Kill' command in `xsbug`

> Note: You must force-quit the simulator to tear down the test setup

phoddie, 2019-09-23:
@zgramana - nice example. And, yes, it hangs. :(

The fix seems straightforward. I tested on macOS. It should work everywhere. Would you give it a try? In `audioOut.c` in the function `xs_audioout_enqueue` move lines [648 to 652](https://github.com/Moddable-OpenSource/moddable/blob/78caaabe6ead2eac7ba82a20a8e755ba948f1e6d/modules/pins/i2s/audioout.c#L648-L652) to directly after the unlock in lines [654 - 662](https://github.com/Moddable-OpenSource/moddable/blob/78caaabe6ead2eac7ba82a20a8e755ba948f1e6d/modules/pins/i2s/audioout.c#L654-L662). Please let me know how that goes.

We'd be happy to accept the PR. You'll just need to complete the [CLA](https://github.com/Moddable-OpenSource/moddable/blob/78caaabe6ead2eac7ba82a20a8e755ba948f1e6d/licenses/Moddable%20CLA%20-%20individual.pdf).

zgramana, 2019-10-21:
@phoddie The fix works when tested on a Moddable Two and a Mac via the Simulator. Cheers!

- [ ] Do you want the change to `audioOut.c` added to the PR?
- [ ] Where does the CLA PDF go once signed?

phoddie, 2019-10-21:
Thanks for confirming. I committed the change already so it is in the current tree.

You can email the executed CLA PDF to info@moddable.com. Thank you!

zgramana, 2019-10-23:
- [x] Email signed CLA

closed 2020-08-28 by phoddie