Moddable-OpenSource/moddable #264: ESP32 windows - ming windows closing before error can be read

wilberforce, 2019-09-15:
the esptool step is outputting an error message, unfortunately it flashes too quick before it can be read....

With a cmd prompt you can add a `pause` statement to wait for a key - how can this be added to the ming build and which script needs to be modified?

Thanks.

bfriedkin, 2019-09-16:
@wilberforce -

The file you want to modify is %MODDABLE%\tools\mcconfig\nmake.esp32.mk.

What I usually do is add a sleep statement to pause the build long enough so that the error can be read. To pause the MinGW window during the first "defconfig" step, add a 30 second sleep command to the end of line 290:

```
if exist $(TMP_DIR)\_s.tmp ($(MSYS32_BASE)\msys2_shell.cmd -mingw32 -c "echo Running GENCONFIG...; BATCH_BUILD=1 DEBUG=$(DEBUG) IDF_BUILD_DIR=$(IDF_BUILD_DIR_MINGW) SDKCONFIG_DEFAULTS=$(SDKCONFIG_FILE_MINGW) make defconfig; sleep 30s")
```

If that isn't where the problem happens, you can do something similar at the problem point in line 265.

Regards,
Brian

wilberforce, 2019-09-16:
Thanks!

Cheers Rhys

wilberforce, 2019-09-18:
@bfriedkin

The `make flash` was failing as `UPLOAD_PORT` was not defined and it defaults to a Linux tty.

Perhaps a warning should be given if this is not defined?

bfriedkin, 2019-09-18:
@wilberforce -

Good to hear you were able to track down the issue. I understand it is a tricky problem to catch, given how the ESP32 build opens a MinGW shell that closes automatically. I've updated the `nmake.esp32.mk` file to check that the required environment variables are defined. The update will be available in our next open source push.

Regards,
Brian

wilberforce, 2019-09-18:
@bfriedkin 

very nice.

closed 2019-09-18 by wilberforce