Moddable-OpenSource/moddable #268: "Break: (host): dead strip!" in tricolor-led example

jernst, 2019-09-22:
Attempting to run the `tricolor-led` example on the Moddable Two with `mcconfig -d -m -p esp32/moddable_two`. The debugger stops at line 5 of file `moddable/build/tmp/esp32/moddable_two/debug/tricolor-led/mc.config.js` with "Break: (host): dead strip!":
```
/* WARNING: This file is automatically generated. Do not edit. */

import Screen from "ili9341";
import Touch from "ft6206";
export default Object.freeze({
	Screen,
	Touch,
	"r_pin": 25,
	"g_pin": 26,
	"b_pin": 27,
	"format": "RGB565LE",
	"rotation": 0,
}, true);
```
What might be the problem here?

phoddie, 2019-09-24:
Good find.

The problem is that the implementation of `Object.freeze` has been stripped because the linker determined it was unused at runtime. That is because `mc.config.js`is intended to be executed during preload, not runtime. The fix is to force `mc.config.js` into the preload list. In `mcmanifest.js` add the following [after line 1091](https://github.com/Moddable-OpenSource/moddable/blob/public/tools/mcconfig.js#L1091).

```
this.preloads.push("mc/config.xsb");
```

You'll need to rebuild tools after making the change. We'll get this into the next update.

phoddie, 2019-10-11:
Fix committed.

closed 2019-10-11 by phoddie