Moddable-OpenSource/moddable #270: XS build

cup, 2019-09-26:
I am interested in a build of XS:

https://github.com/Moddable-OpenSource/moddable/tree/public/xs

but I dont see one here:

https://github.com/Moddable-OpenSource/moddable/releases

phoddie, 2019-09-27:
> I am interested in a build of XS 

Thank you for your interest.

XS is a JavaScript engine. It must be hosted in some app to be used. The Moddable SDK contains instructions for building a variety of applications that use the XS JavaScript engine. The [Getting Starting](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md) document is a good place to start.

If you want to experiment with the XS JavaScript engine alone, you can run scripts using the `xst`command line. You can [build](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/xst.md) `xst` from the sources. We also provide[ binary releases of xst](https://github.com/Moddable-OpenSource/moddable-xst/releases). If you are already using [jsvu](https://github.com/GoogleChromeLabs/jsvu), it can fetch the latest version of `xst` for you.

cup, 2019-09-27:
@phoddie that is great, thank you

closed 2019-09-27 by cup