Moddable-OpenSource/moddable #273: SSL issue with /examples/network/services/twilio/main.js

noelportugal, 2019-10-06:
I'm using a Moddable Two (esp32/moddable_two)
After using:
`
mcconfig -d -m -p esp32/moddable_two ssid="XXXX" password="XXXX"
`
I get the following error:
```
Wi-Fi connected to "XXXXXX"
IP address 192.168.X.XXX
got time
> clientHello.packetize
>> handshakeProtocol.packetize
>>> recordProtocol.packetize
>>>> tlsPlainText.packetize
>>>>> tlsCompressed.packetize
>>>>>> tlsCipherText.packetize
> recordProtocol.unpacketize
>> tlsCipherText.unpacketize
>>> tlsCompressed.unpacketize
>>>> tlsPlainText.unpacketize
>>>>> handshakeProtocol.unpacketize
>>>>>> serverHello.unpacketize
> recordProtocol.unpacketize
>> tlsCipherText.unpacketize
>>> tlsCompressed.unpacketize
>>>> tlsPlainText.unpacketize
>>>>> handshakeProtocol.unpacketize
>>>>>> certificate.unpacketize
/Users/[xxx]/[xxx]/[xxx]/moddable/modules/files/resource/Resource.c (43) # Break: Resource: Resource not found: ca106.der!
SMS send HTTP request failed
```

I did enabled trace:
```
let request = new Request({
	host: "api.twilio.com",
	path: `/2010-04-01/Accounts/${account}/Messages.json`,
	method: "POST",
	headers: [	"content-type", "application/x-www-form-urlencoded",
				"content-length", body.length,
				"Authorization", "Basic " + Base64.encode(account + ":" + token)],
	body: body,
	response: String,
	port: 443,
	Socket: SecureSocket,
	secure: {
		trace: true
	}
});
```

lprader, 2019-10-07:
That error means the TLS certificate `ca106.der` is missing. I'll fix that in the example, but it's an easy fix if you don't want to wait until our next open source push. To add a TLS certificate, you just include it in the `resources` object in the manifest. In this case:

```
     ...
    "resources": {
        "*": [
            "$(MODULES)/crypt/data/ca106",
        ]
    },
    ...
```
[This blog post](https://blog.moddable.com/blog/https/) has some more information (in the **Manifest** section near the bottom), if you're interested.

lprader, 2019-10-07:
Oh...one more thing. I tried running the example with the change above, and the HTTP response said "Invalid TLS version". You'll have to change that as well. You can upgrade the TLS version using the `secure` object in your HTTP `Request`. For TLS 1.2:

```
secure: { protocolVersion: 0x303 },
```

Clearly we haven't run this example in a while ;) Thanks for catching that.

noelportugal, 2019-10-07:
Awesome! This works.

closed 2019-10-07 by noelportugal