Moddable-OpenSource/moddable #274: Please explain 'import {} from "piu/MC";'

mikew67, 2019-10-21:
Appears in several of your examples.  Suspect I get the general idea, but wasn't able to track down the details anywhere in your docs (or with Google in the wider JS or ES world.

- Is "{}" the same as "*" in this context?

- What does "MC" mean?  No file with that name in the piu directory.  Some special meaning?  Described where?  Just a coincidence that your tools start with "mc..."?

In any case, would like to see an explanation of this In the docs.  Not sure where.

mikew67, 2019-10-22:
OK, I see now that there's a "pic/MC" in the **modules** directory of the repo, so it looks like this import statement imports all the modules in that sub-directory.  But, for completeness, is there documentation that the import statement will, by default, search the modules directory?  Any other directories?  Is it in the mcconfig documentation.

Thanks.

Mike

mikew67, 2019-10-22:
Digging deeper, is it necessary to include the following line in the Balls example code:
`import {} from "piu/MC`

Because the manifest for balls includes:
`"include": [
		"$(MODDABLE)/examples/manifest_base.json",
		"$(MODDABLE)/examples/manifest_piu.json",
	],`

and manifest_piu.json includes:
`	"modules": {
		"commodetto/Bitmap": "$(COMMODETTO)/commodettoBitmap",
		"commodetto/Poco": "$(COMMODETTO)/commodettoPoco",
		"commodetto/*": "$(COMMODETTO)/commodettoPocoBlit",
		"commodetto/cfe": "$(COMMODETTO)/cfeBMF",
		"commodetto/parseBMP": "$(COMMODETTO)/commodettoParseBMP",
		"commodetto/parseRLE": "$(COMMODETTO)/commodettoParseRLE",
		"piu/*": "$(PIU)/All/piu*",
		"piu/CombTransition": "$(PIU)/MC/piuCombTransition",
		"piu/MC": "$(PIU)/MC/piuMC",
		"piu/WipeTransition": "$(PIU)/MC/piuWipeTransition",
		"*": [
			  "$(PIU)/MC/piuDie",
			  "$(PIU)/MC/piuFont",
			  "$(PIU)/MC/piuImage",
			  "$(PIU)/MC/piuRegion",
			  "$(PIU)/MC/piuTexture",
			  "$(PIU)/MC/piuView",
		]
	},
`

Note: was getting ready to try this, but have to change my setup because Apple changed the default shell in Catalina to zsh.  See my next issue on this.

phoddie, 2020-08-28:
The use of JavaScript here is standard and the manifest [defines the module mapping](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/tools/manifest.md#modules) as usual. Some details follow.

The statement

```
import {} from "piu/MC
```

imports the `piu/MC` module defined in the [manifest_piu.json](https://github.com/Moddable-OpenSource/moddable/blob/276e962b1106b798b388663f0f311bc32f917c23/examples/manifest_piu.json#L11)

```
"piu/MC": "$(PIU)/MC/piuMC",
```

referencing the file [piuMC.js](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/piu/MC/piuMC.js)

The `import` statement here does not reference any symbols exported by the module, but does cause the module to be loaded and run, which has side effects, specifically initializing several global variables. [MDN](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Statements/import) explains "(importing) a module for its side effects only" as follows:

> Import an entire module for side effects only, without importing anything. This runs the module's global code, but doesn't actually import any values.

The use of "MC" in the Moddable SDK is generally an abbreviation for MicroController. In the case of Piu, for example. piuMC refers to the version of Piu for resource constrained microcontrollers contrasted with piuPC, the version of Piu for Personal Computers (used, for example, to implement xsbug).

closed 2020-08-28 by phoddie