Moddable-OpenSource/moddable #280: Build of simple Pins/Blink example app for ESP32 is broken & xsbug behaves weirdly, too

mikew67, 2019-11-02:
Getting really weird behavior.  Here's a minimal repeatable sequence:

1. Running on a MacBook Pro 10,1 (15" Retina, 2013), using Catalina 15.1 (final)
2. Target is an ESP32_Devkitc_V4 (Had trouble getting serial USB connection to Moddable Two -- see the next issue statement.)
3. Did a fresh Mac reboot; started Terminal, built, loaded and ran Hello World example. All OK. (See attached Terminal log and screenshot of xsbug.). Note in particular pathname of file open in xsbug view pane and console output.
[image]
4. Killed xsbug (so it'll start fresh in next step).
5. Built, loaded and ran Pins/Blink example app.  From Terminal log, everything worked fine, but...
[image]
6. xsbug is doing something completely different: (a) its source view is still open on Hello World, not blink, and (b) its console log (and apparently the code that got loaded) is entirely different than Blink.
[image]

Here's the full Terminal log (including the contents of main.js and the manifest file from Blink):
[Blink debug log.txt](https://github.com/Moddable-OpenSource/moddable/files/3800973/Blink.debug.log.txt)

And a check of some Environment Variables:
[image]

What's going on here?  Some boneheaded configuration error on my part?

TIA,

Mike

mkellner, 2019-11-02:
It looks like blink is running because of the output. The source display is showing the last opened file. Blink has never yet hit a breakpoint.

Click the "main" tab at the top to show the machine that is running. Then you should see the instrumentation and controls like stop and pause. Press pause.

mkellner, 2019-11-02:
[image]

mkellner, 2019-11-02:
Helloworld automatically stops and displays the source because there is a `debugger;` strategically placed at the top of the file.

mikew67, 2019-11-02:
Aha, thanks, Michael.  Looks like a case of RTFM -- specifically, xsbug in my case.  Also, didn't think the app was running because I didn't see any "flashing LEDs", understandable once I double-checked the schematic for the ESP32_Devkitc_V4 and realized the only LED was a power indicator, not a GPIO-connected indicator. Hung a logic probe on GPIO2.  Blinking away  :-)
Thanks for the quick reply.
Mike

closed 2019-11-02 by mikew67