Moddable-OpenSource/moddable #282: Constructor that returns an async IIFE fails in XS

dtex, 2019-11-03:
Hi,

I have the following pattern that works in V8 but is throwing in XS.

````js
class Thing {  
  constructor(myName) {
    return (async () => {
      this.name = myName;
      return this;
    })();
  }
}

let thing1 = new Thing("Hartman");
let thing2 = new Thing("Lovitz");
````

The message I see in the xsbug console is
````bash
/Users/donovan/Projects/embeddedTest/main.js (4) # Break: ?: new async!
````

I understand that this example isn't actually doing anything that requires async, I just wanted to provide a minimal example.

dtex, 2019-11-03:
Oh wait, I think I found it:  

From Conformance.md
> ... some platforms do not support eval, new Function, new AsyncFunction or new Generator

I'm guessing ESP8266 does not support new AsyncFunction

dtex, 2019-11-04:
Is there a make file or something I can edit to enable new AsyncFunction on ESP8266? I'd like to see what impact it has on my use case.

phoddie, 2019-11-04:
@dtex , thanks for the report.

The ESP8266 host does support AsyncFunctions. The note in the conformance document is a bit misleading as it refers to the fact that we usually strip `eval` (and related functions such as `AsyncFunction` from the ESP8266 target. While `eval` and friends are not present by default, they do work there. But, your example does not depend on `eval`.

Your code does work if you change the name `constructor` to `foo` and then call `thing1.foo("Radner")` in the test. The problem appears to be specific to using the `async` keyword inside a constructor. 

We'll look into the details.

phoddie, 2019-11-04:
@dtex, the error is caused by some obsolete code to prevent `new`  from being called on async functions. Your code works if that check is disabled. To try that, comment out lines 919 and 920 in [xsRun.c](https://github.com/Moddable-OpenSource/moddable/blob/public/xs/sources/xsRun.c#L919-L920).

We are working on an update to XS. We'll include this fix in the update.

Thank you,

dtex, 2019-11-04:
I'll try it out tonight. I'm excited because this is the pattern I wanted to use. I had been exploring alternatives and none of them were quite as nice.

Thanks so much!

dtex, 2019-11-05:
Thanks, I'm back on track. 

Commenting out those lines worked and I will keep an eye out for the next version.

closed 2019-11-05 by dtex