Moddable-OpenSource/moddable #287: Building for the ESP32_Devkitc_V4 vs Moddable Two with the Moddable SDK

mikew67, 2019-11-13:
Folks,

Building for your Moddable Two with your SDK works great, especially easy with the explicit Getting Started docs.  I've discussed my use of this environment for my educational robotics project before.  At the last scripting meetup, it was mentioned that the ESP32_Devkitc_V4 would also be appropriate for the HW platform.  Nice suggestion.  Smaller, cheaper, more pins accessible before resorting to fly wires, and the display isn't needed.  But I'm having some difficulties with uploading code to the Devkit, and this seems to be a common problem from discussions online.

Here's my environment:

- MacBook Pro (10,1; 2013), running macOS Catalina 10.15.2 beta, with your current build environment and recommended version of the ESP IDF, etc.

Here's a **comparison of the two build targets and how your SDK treats them**:

- Platform ID: esp32/moddable_two vs esp32

- /dev/cu.*: /dev/cu.usbserial-0001 vs /dev/cu.usbserial-14120 (Note: Neither one is /dev/cu.SLAB_USBtoUART)

- USB-to-serial chip: CP2102 vs CP2102

- Driver loaded by macOS: DriverKit.AppleUSBSLCOM (v1) vs. DriverKit.AppleUSBSLCOM (v1) (Note: Neither one loads the Silicon Labs 21xx driver, although I believe it to be installed.)

- Driver ID: com.apple.DriverKit.AppleUSBSLCOM vs com.apple.DriverKit.AppleUSBSLCOM

- **Load log for Moddable-generated binary for Moddable Two**:

`esptool.py v2.6`
`Flashing binaries to serial port /dev/cu.usbserial-0001 (app at offset 0x10000)...`
`esptool.py v2.6`
`Serial port /dev/cu.usbserial-0001`
`Connecting....`
`Chip is ESP32D0WDQ5 (revision 1)`
`Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None`
`MAC: 24:6f:28:19:b5:84`
`Uploading stub...`
`Running stub...`
`Stub running...`
`Changing baud rate to 921600`
`Changed.`
`Configuring flash size...`
`Auto-detected Flash size: 4MB`
`Compressed 16656 bytes to 10790...`
`Wrote 16656 bytes (10790 compressed) at 0x00001000 in 0.2 seconds (effective 789.4 kbit/s)...`
`Hash of data verified.`
`Compressed 424032 bytes to 248035...`
`Wrote 424032 bytes (248035 compressed) at 0x00010000 in 4.0 seconds (effective 857.0 kbit/s)...`
`Hash of data verified.`
`Compressed 3072 bytes to 120...`
`Wrote 3072 bytes (120 compressed) at 0x00008000 in 0.0 seconds (effective 2130.0 kbit/s)...`
`Hash of data verified.`

**Upload works fine. Note the baud rate: 921600**

- **Load log for Moddable-generated binary for Devkit**:
`esptool.py v2.6`
`Flashing binaries to serial port /dev/cu.usbserial-14120 (app at offset 0x10000)...`
`esptool.py v2.6`
`Serial port /dev/cu.usbserial-14120`
`Connecting........_____....._____....._____....._____....._____....._____....._____`
`  `
`A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header`
`make[1]: *** [flash] Error 2`

**Upload failed.** 

Don't understand why the connection attempt to the Devkit fails.  A serial terminal emulator program, Serial, sees no difference between the two (as noted in the identical chip and driver data above). Also the upload to the Devkit still failed when retrying with Boot button held down either before or during connection.

Comments?  Do I have bad Devkit HW? (Tried two.). How do I fix this?

TIA,

Mike

mikew67, 2019-11-13:
Hmm... interesting.  Using the Moddable Two board (instead of the ESP32_Devkitc_V4) at the moment.  Added some wires (e.g., capacitance and noise) and now the M2 is failing to connect **sometimes**, too.
Perhaps this is a HW issue.

ederle, 2019-11-13:
hello,

Same error for me:
```
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
make[1]: *** [flash] Error 2
```
I changed baud rate (UPLOAD_SPEED) to 115200 and it works fine.

cheers

mikew67, 2019-11-13:
Where do you change that? Environment variable or setting in the SDK code?

Thanks.

ederle, 2019-11-13:
in ***moddable/tools/mcconfig/make.esp32.mk***

Change :
```
UPLOAD_SPEED ?= 921600
DEBUGGER_SPEED ?= 460800
```
to
```
UPLOAD_SPEED ?= 115200
DEBUGGER_SPEED ?= 115200
```
I hope that it will work for you.

mikew67, 2019-11-13:
Hmm... Three tries at 115200.  No go.  Will try again later.

mikew67, 2019-11-13:
Here's a recommended HW fix.  Not sure I'll try it.  It purports to replace "holding down the boot button", which I tried and doesn't work:
https://randomnerdtutorials.com/solved-failed-to-connect-to-esp32-timed-out-waiting-for-packet-header/

wildclocks, 2019-11-13:
Press and hold that boot button when it's doing this:

> Connecting........____

Let go after it starts downloading. This usually works for me when I'm having problems with one of those modules catching the download.

I've also had a variety of issues with USB on Mac after quite some time of many devices plugging and unplugging into USB. Rebooting usually resolves that issue (ie. when you couldn't the Moddable Two)

mikew67, 2019-11-14:
Even tried removing C15 per instructions [here](https://www.esp32.com/viewtopic.php?t=5647), to no avail.  Still won't boot, even at 56K baud.

mikew67, 2019-11-14:
OK, gang, I appear to have the **issue of connecting to ESP32Devkitc_V4 boards** (as well as the Moddable Two, actually anything with a Silicon Labs CP2102N UART chip) **solved for macOS Catalina**.

Note that Apple is in the midst of changing their driver interface for macOS, hoisting the driver code from **.kext** files in privileged system space, to **.dext** files in user space, a _good thing_ long term, but causing turbulence in the short term.  And it's not completed yet.

First of all, here are the **things that don't work:**

- Holding down the Boot button during connection

- Lowering the serial speed of the UART link

- Removing the C15 capacitor on the ESP32Devkitc_V4 board

Here's what I got to work:

- **Updated my macOS Catalina install to 15.2 beta 2** (i.e., just released in the last day or two). Note that this totally breaks access to CP2102N devices, because the macOS upgrade moves to a "built-in" driver for this device.  I got erratic behavior then, sometimes the USB device would show up on the System Report or not, and sometimes it would show up in a "ls -al /dev/cu.*" and sometimes not.  Here's a screenshot of what the excellent Serial app would show for driver info:
[image]

- After the macOS upgrade, reinstalled the latest [Silicon Labs driver, V. 5.2.2](https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers). 
[image]

Note the difference in driver/kext used.

Now both Moddable Two and ESP32Devkitc_V4 cards connect just fine, even with serial speeds returned to default.

Issue closed for me (for now -- until Apple changes something again :-) I'll leave it open for a couple days for others to comment and then close it.

mikew67, 2019-11-23:
OK, my access to /dev/cu.SLAB_USBtoUART has persisted, even through the macOS Catalina 15.2 beta 3 update, so I'm thinking my .kext's are happy.  Closing this issue.

closed 2019-11-23 by mikew67