Moddable-OpenSource/moddable #290: Order of import statements question

dtex, 2019-11-21:
The following code works fine:

````js
import { inOutQuad } from "j5e/easing";
import Led from "j5e/led";

(async function() {
	const led = await new Led({ pin: 12, io: "builtin/pwm"});
	led.pulse({
		duration: 2000,
		easing: inOutQuad
	});
})();
````

But if I change the order of the import statements, it throws ```get inOutQuad: not initialized yet!```:

````js
import Led from "j5e/led";
import { inOutQuad } from "j5e/easing";

(async function() {
	const led = await new Led({ pin: 12, io: "builtin/pwm"});
	led.pulse({
		duration: 2000,
		easing: inOutQuad
	});
})();
````
Is that expected behavior?

phoddie, 2019-11-22:
At first glance, it looks like a problem. The import statements should be completed before the code in the body of the module is executed. But maybe there's something else going on.  Would it be possible for you to narrow down the example to  something small and reproducible outside your code base?

dtex, 2019-11-22:
Ya, I'll get something to you this weekend.

dtex, 2019-11-24:
I'm having a heck of a time isolating this. The behavior is so erratic that I was beginning to believe I was encountering some resource limitations on the ESP8266, but I've been able to delete whole swaths of code and still have it fail, and just changing the import order makes it work. One misleading thing in my code above though. There was a 3rd module that I didn't think was relevant so I removed it for simplicity, but it turns out it made a difference after all. I'm going to create a dependency map and see if I can trace out some potential culprit in the way I'm referencing internal dependencies. Maybe I've got something loopy going on.

dtex, 2019-11-24:
Increasing ```creation.heap.initial``` makes it work. As far as I'm concerned, that solves it, I just think it's weird that import order also solves it. I'm sure if I understood the creation properties better it would all make perfect sense.

phoddie, 2020-10-08:
Importing modules uses memory, so rearranging the import order could change the error behavior. The best solution is to preload the modules, because then importing them takes no memory.

I'm going to close this issue out. If there's still something to look at, please reopen it. Thank you.

closed 2020-10-08 by phoddie