Moddable-OpenSource/moddable #292: Is the build process in the current repo broken?

mikew67, 2019-11-23:
(Note: I just did a pull on your repo and got commits through two days ago by Patrick.  Was that a mistake?)

Or did I just muck up my local copy of the repo with additional files as described in issue 291?

Even for your simple blink example, I'm getting: 

> mike@Mikes-MacBook-Pro-10-1 blink % mcconfig -d -m -p esp32            
> \# cc modTimer.c.o
> /var/folders/hp/_vddx6wx4ws6n5wsbb8793l80000gn/T//ccxZ17Ko.s: Assembler messages:
> /var/folders/hp/_vddx6wx4ws6n5wsbb8793l80000gn/T//ccxZ17Ko.s:660: Warning: setting incorrect section attributes for .rodata
> ...

at the beginning of the build .

xsbug says: 
>\# Break: (host): invalid hint!

and I drop into GDB on the terminal, rather than xsbug.

What's up?  I'm lost.

andycarle, 2019-11-23:
Hi Mike,

Most likely you just need to do a clean build of the Moddable SDK tools.

Just go in to $MODDABLE/build and delete the `bin` and `tmp` directories. Then follow the platform-appropriate instructions in the [Getting Started Guide](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md) to rebuild the tools. For instance, you would start at Step 5 of the macOS host environment setup instructions or Step 6 for Windows.

Please let me know if that doesn't work for you.

&nbsp;- Andy

mikew67, 2019-11-23:
OK, back in action again after trashing my local copy of the repo (after saving my local additions :-) and going back through the sequence in "Getting Started".

So we have a "Getting Started" doc; but now we need a "Keeping current" doc that gives us hints on how to stay current with the repo, and when to rebuild the tools, etc.  

Thanks,

Mike

mikew67, 2019-11-23:
Sorry.  Have to reopen this.  Still getting errors in weird places.

- Here's an error right at the beginning of the build of my app:

`mike@Mikes-MacBook-Pro-10-1 T3-PWM % mcconfig -d -m -p esp32
\# xsc main.xsb
\# xsl modules
\# cc mc.xs.c (slots in flash)
/var/folders/hp/_vddx6wx4ws6n5wsbb8793l80000gn/T//ccY8biSI.s: Assembler messages:
/var/folders/hp/_vddx6wx4ws6n5wsbb8793l80000gn/T//ccY8biSI.s:3183: Warning: setting incorrect section attributes for .rodata.mod.1
\# ld xs_esp.bin
...
`

- Then during execution, both Digital and PWM object creation are causing errors:
`.../modules/pins/digital/digital.c (75) # Break: Digital: can't init pin!`
and
`.../modules/pins/pwm/esp32/pwm.c (100) # Break: PWM: configure led channel failed!`

Don't know how to get beyond these errors.
Source code attached:

[T3-PWM.zip](https://github.com/Moddable-OpenSource/moddable/files/3882144/T3-PWM.zip)

mkellner, 2019-11-23:
I built the app you uploaded and didn't have a problem.

In your comment above, the first bullet suggests an error. Please note that it is a "Warning".

I'm guessing that your build had pieces of a previous build. Please try
   `rm -rf $MODDABLE/build/bin/esp32 $MODDABLE/build/tmp/esp32`
to remove the temporary build files and build output for **esp32**.

If you need to rebuild the tools too, then remove the entirety of the $MODDABLE/build/bin and $MODDABLE/build/tmp files. You will have to rebuild the tools by
    `cd $MODDABLE/build/makefiles/mac ; make`

Then return to your application directory and issue the `mcconfig`.

mikew67, 2019-11-23:
Many thanks for the responses:

- Did the tools rebuild again (per Andy's suggestion)

- Then removed my app's temp build files per Michael's instructions

- Build of my T3-PWM code went fine, but...

- When I stepped through it with xsbug, it failed at the first creation of a new Digital object (see screen shot below) with pin 36 and mode Output (=>8, apparently).
Any advice from this point on?  Should I checkout a _more stable_ version of the repo? What date and/or commit id?  

I'm using esp32 as the target platform.  GPIO 36 is a valid GPIO number.  Am I getting hung up on some hidden config setting?
[image]

BTW, both Andy and Michael's advice should be captured in a "Keeping Going" addendum to the Getting Started doc!

Thanks, as always,

Mike

mikew67, 2019-11-23:
Bingo.  Tried 5 for the GPIO number and the new Digital call went through OK.  Looks like I have to modify some config settings for my ESP32_Devkitc_V4 board.  Now, let me see, where are those board config files....

mikew67, 2019-11-23:
Sigh... Not out of the woods yet.  Getting panic interrupts into GDB on the "new Digital" call, even with a clearly legal pin.  Will have to check this out further later today.  Other tasks take priority right now....
Mike

mikew67, 2019-11-23:
Is something missing from my local repo?  The symbol, GPIO_NUM_MAX, appears in only one place (clearly an error) in my moddable directory, in:

> ${MODDABLE}/modules/pins/digital/esp32/modGPIO.c; Line 37

in the following code snippet:

`int modGPIOInit(modGPIOConfiguration config, const char *port, uint8_t pin, uint32_t mode)
{
	int result;

	if ((pin > GPIO_NUM_MAX) || port) {
		config->pin = kUninitializedPin;
		return -1;
	}
...
`

I'll rebuild my entire moddable environment from the current repo and check to see if this is still the case.  If so, I'll back up to the repo as of a month ago(?)

Any other suggestions?

Mike

mikew67, 2019-11-23:
Reloaded current repo; GPIO_NUM_MAX is undefined (only occurs where it's used in modGPIO.c).  Clearly a **bug**.  And this is why my calls to create "new Digital" objects was failing, but direct method calls of the class object, Digital, in the Blink example:
`Digital.write(2, ~count & 1);`
wasn't.
Backtracking in the repo to find a stable place starting with the Meetup date last month...
Mike

mkellner, 2019-11-24:
In Espressif's SDK, the file esp-idf/components/driver/include/driver/gpio.h
should contain GPIO_NUM_MAX as it is a hardware dependent value defined by the Espressif.

There should be an environment definition IDF_PATH which contains the path to the SDK.

Note that for other platforms, IDF_PATH is probably irrelevant as they have their own SDKs and definitions.

mikew67, 2019-11-24:
Michael,

Thanks.  The missing definition(?) of GPIO_NUM_MAX is probably irrelevant.  The logic to check for _legal_ GPIO PIN number must be happening somewhere else and with more smarts than the modGPIO.c file.  That check is creating an init error because GPIO 39 is **input only**.  According to paragraph 4.1 of the ESP32 tech ref:

> Note that the I/O GPIO pads are 0-19, 21-23, 25-27, 32-39, while the output GPIOs are 0-19, 21-23, 25-27, 32-33. _GPIO pads 34-39 are input-only_. (My emphasis.)

If I change my GPIO number from 39 to 33, the init as an output works (e.g., using Blink as a test).

mikew67, 2019-11-24:
OK, closing out this issue.  In the final analysis:

- Root couse was "pilot error" again :-) -- trying to use a GPIO pin for output that was for input only, confounded by a not very helpful error message and not especially obvious hardware documentation.

- Chasing this down would have been easier with better access to gdb.  Can anyone explain how to use gdb in the context of xsbug?  Since I'm certainly going to be mixing low-level C code with JS, knowing how to use gdb in the Moddable environment would be very useful.  Any volunteers for a quick tutorial?  Anybody up for adding a gdb section to the XS documentation?

As always, many thanks to the team, especially Andy and Michael for getting me out of the ditch again!

Mike

-

closed 2019-11-24 by mikew67