Moddable-OpenSource/moddable #293: Hardcoded ~/esp32 path for xtensa-esp32-elf-gcc executable

mgcrea, 2019-11-26:
Was trying the getting started examples for the esp32 platform

Wanted to reuse my working platformio esp-idf setup:

```bash
export UPLOAD_PORT=/dev/cu.SLAB_USBtoUART
export IDF_PATH="$HOME/.platformio/packages/framework-espidf"
export MODDABLE="$HOME/Developer/moddable"
export PATH="${MODDABLE}/build/bin/mac/release:$HOME/.platformio/packages/toolchain-xtensa32/bin:$PATH"
```

However it looks like the `xtensa-esp32-elf-gcc` binary path is hardcoded:

```bash
$ mcconfig -d -m -p esp32
# cc xsHost.c (strings in flash)
make: /Users/olivier/esp32/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: No such file or directory
make: *** [/Users/olivier/Developer/moddable/build/tmp/esp32/debug/lib/xsHost.c.o] Error 1
```

```bash
$ which xtensa-esp32-elf-gcc
/Users/olivier/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc
```

Workaround:

```bash
ln -s $HOME/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc $HOME/esp32/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
```

bfriedkin, 2019-11-26:
@mgcrea -

We recommend following the ESP32 setup instructions in our [Getting Started](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md) guide for your development host to ensure compatibility with the Moddable build. That said, our [ESP32 make file](https://github.com/Moddable-OpenSource/moddable/blob/public/tools/mcconfig/make.esp32.mk) uses a `TOOLS_ROOT` environment variable that points at the directory containing the toolchain. You can likely override that variable in your environment instead of using a symbolic link:

```
export TOOLS_ROOT=/Users/olivier/.platformio/packages/toolchain-xtensa32
```

Regards,
Brian

PaulFreund, 2019-12-12:
Starting with the 4.0 version of ESP-IDF there are two script included, esp-idf/install.sh and esp-idf/export.sh which manage all the dependencies and neccessary paths. TOOLS_ROOT will change with every version (TOOLS_ROOT=~/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf currently). export.sh automatically puts the compiler bin folder in the PATH which I think is better than using a fixed TOOLS_PATH, is there any reason for that?

Also with the current version esp_task.h is not found which I will investigate now

closed 2022-02-10 by phoddie