Moddable-OpenSource/moddable #294: Issue when uploading to a TTGO T-Koala board

mgcrea, 2019-11-26:
Looks like it should work outside the box but the bundled esptool.py call does not work for my [TTGO T-Koala board](http://www.lilygo.cn/prod_view.aspx?TypeId=50033&Id=1125&FId=t3:50033:3):

```bash
esptool.py v2.6
Flashing binaries to serial port /dev/cu.usbserial-1410 (app at offset 0x10000)...
esptool.py v2.6
Serial port /dev/cu.usbserial-1410
Connecting.....
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: b4:e6:2d:de:cb:69
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...

A fatal error occurred: Invalid head of packet (0xE0)
make[1]: *** [flash] Error 2
```

Also encountered:

```bash
Configuring flash size...

A fatal error occurred: Timed out waiting for packet header
```

When using platformio to target the same board, it works:

```bash
esptool.py v2.6
Serial port /dev/cu.usbserial-1410
Connecting......
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: b4:e6:2d:de:cb:69
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 15872 bytes to 10319...

Writing at 0x00001000... (100 %)
Wrote 15872 bytes (10319 compressed) at 0x00001000 in 0.3 seconds (effective 489.6 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.0 seconds (effective 2480.2 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 11036.6 kbit/s)...
Hash of data verified.
Compressed 195504 bytes to 95706...

Writing at 0x00010000... (16 %)
Writing at 0x00014000... (33 %)
Writing at 0x00018000... (50 %)
Writing at 0x0001c000... (66 %)
Writing at 0x00020000... (83 %)
Writing at 0x00024000... (100 %)
Wrote 195504 bytes (95706 compressed) at 0x00010000 in 2.4 seconds (effective 655.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
======================================================================================================= [SUCCESS] Took 9.45 seconds =======================================================================================================
```

Here is the platformio config that is working:

```json
$ cat ~/.platformio/boards/ttgo-t-koala.json 
{
  "build": {
    "core": "esp32",
    "extra_flags": "-DARDUINO_T-Koala",
    "f_cpu": "240000000L",
    "f_flash": "40000000L",
    "flash_mode": "dio",
    "ldscript": "esp32_out.ld",
    "mcu": "esp32",
    "variant": "esp32"
  },
  "connectivity": [
    "wifi",
    "bluetooth",
    "ethernet",
    "can"
  ],
  "debug": {
    "default_tool": "ftdi",
    "onboard_tools": [
      "ftdi"
    ],
    "openocd_board": "esp32-wrover.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "TTGO T-Koala",
  "upload": {
    "flash_size": "4MB",
    "maximum_ram_size": 327680,
    "maximum_size": 4194304,
    "protocols": [
      "esptool",
      "espota",
      "ftdi"
    ],
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "http://www.lilygo.cn/prod_view.aspx?TypeId=50033&Id=1125&FId=t3:50033:3",
  "vendor": "TTGO"
}
```

Is there any way to change the upload speed?
Is it possible to perform a manual upload (did not see the actual command in the logs).

andycarle, 2019-11-28:
Hi Olivier,

The environment variable for changing the upload speed of the ESP32's `make flash` command is `ESPBAUD`. So, you should be able to do `ESPBAUD=460800 mcconfig -d -m -p esp32 -v`.

Give that a shot and see if it works any better with your board. If not, please do let me know and we'll see if we can sort it out.

&nbsp;- Andy

mgcrea, 2019-11-29:
Works, thanks!

closed 2019-11-29 by mgcrea