Moddable-OpenSource/moddable #295: xsbug fails to open target file and break

mikew67, 2019-11-27:
Folks,

There must be something really strange about my configuration.  Even with the stock Hello World example (with a "debugger;" line in it), the xsbug app opens on my MacBook Pro (running Catalina 15.2 beta 3) but **fails both to open the target main.js file and to pause with a break at the beginning**.  I have done a _complete reinstall_ of the moddable SDK, esp-idf and xtensa-esp32-elf (being careful to get the right version).  Even checked the environment variables and they seem OK.
This change happened suddenly when I was in the midst of developing more complicated code (using Monitor).
Any suggestions?  Imagine xsbug is invoked by some (Python?) script call with the target file path as argument.  Should I be looking for that to debug my problem?

TIA,

Mike (in the ditch again, and it's raining :-)

mikew67, 2019-11-27:
BTW, here are my xsbug prefs:
[image]

mikew67, 2019-11-27:
OK, issue _solved_ -- or at least I can get beyond it, but still need to understand how it happened.

Looks like I have a bad ESP32_Devkitc_V4 board.  Problem doesn't occur with either another board of the same type or a Moddable Two.  Probably mangled some Flash contents (Boot loader?) on the bad board.

- Anybody know how to check and restore the Boot Loader, etc., on one of these boards?

- And how would I know this was the cause?  The build and load log (attached) doesn't seem to indicate any problem.  If xsbug sees a problem, sure would be nice if it put up a warning.
[Bad board build & load log.txt](https://github.com/Moddable-OpenSource/moddable/files/3895103/Bad.board.build.load.log.txt)

andycarle, 2019-11-28:
Hey Mike,

You might try just erasing the entire flash contents on the board you're having trouble with. Sometimes the esptool does seem to get a bit confused about what bootloader should be installed and won't reflash the full thing even when it should.

`esptool.py erase_flash` will do the trick if you have your environment fully set up. If not, you may need to specify a port on the command line, e.g. `esptool.py --port <PORT> erase_flash`

&nbsp;- Andy

mikew67, 2019-11-28:
Andy,

Thanks for the informative reply.  Turns out to have been a hardware problem.  I had a logic probe connected to GPIO0 and that was enough of a load to interfere with the boot loader.  Go figure.  

Mike

closed 2019-11-28 by mikew67