Moddable-OpenSource/moddable #296: What .h to include with xsTrace and xsDebugger?

mikew67, 2019-11-28:
Quick question.  Trying to trace and use xsbug with C code.  Therefore, tried using xsTrace and XSDebugger.  Used "#include xs.h".  But get an undefined function of fxDebugger, etc. 

What file should I include (on the Mac)?

mikew67, 2019-11-28:
And tried xsmc.h but that didn't work either.

mikew67, 2019-11-28:
OK, a bit more detail. xsTrace is a macro that expands as follows:
`xsTrace("my msg"); ==> fxReport(the, "%s", "my msg");`
and it's the first argument, "the", that's undefined and causing a problem.  According to the definition of fxReport, the first argument is of type "xsMachine *", so I presume it should be a pointer to the current XS machine.  How can I fix this?

mikew67, 2019-11-28:
So the convention is to set the variable, “the”, in C code to a pointer to the current machine. But is there an easy way to get this from C? Guess I need to pass it in from the calling JS (since the C code wouldn’t know which machine called it). Right? Unless I hear differently, I’ll proceed on that basis. 
And how do I get it in JS? Some object reference?  Not “this”, but... what?

andycarle, 2019-11-28:
Hi Mike,

I'd recommend taking a look through the [XS in C documentation](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/XS%20in%20C.md) if you haven't already and then experimenting with the [xs-in-c example](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/js/xs-in-c). 

That example app shows how to call into native code from script. The file [rectangle.js](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/js/xs-in-c/rectangle.js) shows the JavaScript language syntax extension used by XS to indicate that a function is implemented in native code (see the [JavaScript @ language syntax extension](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/XS%20in%20C.md#javascript--language-syntax-extension) section of the docs.) In [rectangle.c](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/js/xs-in-c/rectangle.c), you'll see how each JavaScript function implemented in C receives a reference to the current VM as its first argument.

For instance, this function is from that file:
```
void xs_rectangle_get_x(xsMachine *the)
{
	xsRectangle r = xsmcGetHostChunk(xsThis);
	xsmcSetInteger(xsResult, r->x);
}
```
In the context of that function, the `xsTrace("message");` macro would work as you are expecting it to, because `the` is properly set.

&nbsp;- Andy

mikew67, 2019-11-28:
Andy,

Thanks for the quick reply.

> ,,,recommend taking a look through the XS in C documentation...

Yep, did that.

> ...and then experimenting with the xs-in-c example.

Built my touch sensor code starting with the Monitor implementation.  Very similar, just a different sensing mechanism at the pins.

> ...shows how to call into native code from script. 

My API is essentially the same as Monitor's, so that part was easy.

>(see the JavaScript @ language syntax extension section of the docs.)

Aha, here's where I missed the **implied** critical information -- not explicitly stated in the documentation, but evident in the code examples,  the @ syntax for a method call, e.g.,

> get x() @ "xs_rectangle_get_x";

which seems to imply a zero-argument function actually expands into a 1-argument C function call:

> void xs_rectangle_get_x(xsMachine *the)...

Furthermore, the documentation on the xsTrace and xsDebugger macros (and others?) doesn't mention that they depend on a specifically-named variable, "the", being defined properly.

Thanks for getting me straightened out.  But I recommend a couple notes be added to the documentation to help future developers:

**Documentation Recommendations:**

- Add a note to the @ syntax extension section that a call of the form
`get x() @ "xs_rectangle_get_x";`
expands into a one-argument C function call with an added argument:
` void xs_rectangle_get_x(xsMachine *the)`

- Furthermore, both there and at the documentation for xsTrace, xsDebugger, and other VM-dependent utility macros, note the **convention** that they depend on a pointer to the current VM being stored in the variable, "**the**".

Thanks again for the prompt reply that helped me "get my head screwed on in the right direction" and keep me going on this time-critical project.

Mike

phoddie, 2020-10-08:
I don't see how the documentation can be much more clear about the `@` syntax.

- The XS in C documentation contains [a section](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/XS%20in%20C.md#javascript--language-syntax-extension) on the @ syntax.
- It includes working examples of a native constructor, native destructor, and native method calls (with 0 and 1 arguments)

A proposal is welcome.

> Furthermore, both there and at the documentation for xsTrace, xsDebugger, and other VM-dependent utility macros, note the convention that they depend on a pointer to the current VM being stored in the variable, "the".

Added this for the next Moddable SDK update:

> **Note**: The macros in the XS in C API require a reference to the target virtual machine in a variable in the current scope with the name `the` of type `xsMachine *`.

closed 2020-10-08 by phoddie