Moddable-OpenSource/moddable #298: can't get socket, httpget, httpsget examples to work on lin

dckc, 2019-11-29:
I'm  trying to get the httpsget example to work; it just stops at the `debugger` statement on line 84 of `securesocket.js` with `message = -2`.

The httpget example seems to hang.

The socket example hangs for a 2 minutes after logging `Hello, socket.` and then spews a bunch of `SOCKET MESSAGE -2 WITH VALUE undefined`.

I can `curl http://www.example.com/` and `telnet www.example.com 80` just fine.

I'm using version 15c572436bdcb355b6222814f6e8300c0f927700 on Ubuntu 18.04.

bfriedkin, 2019-11-30:
@dckc -

We were able to reproduce the secure socket failure using top-of-tree Moddable SDK on Ubuntu 18.04.1 LTS. Thank you for reporting the issue. The socket module implementation is a bit experimental on desktop Linux as microcontroller targets are our primary focus. But we will take a look and try to sort out the issue.

We weren't however able to reproduce any problems with the socket examples that don't require TLS/SSL, i.e. [httpget](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/network/http/httpget) or [socket](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/network/socket/socket). The Linux socket implementation uses GTK for hostname resolution and I am wondering if that might be failing on your host. Could you give the [dnsresolve](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/network/dns/dnsresolve) example a try and let us know how that goes?

Regards,
Brian

dckc, 2019-11-30:
dnsresolve seems to work fine:

```
"apple.com" resolved to 17.172.224.47
"d" resolved to 2606:4700:10::6816:9dd
"cannot.find.domain" could not be resolved
"moddable.tech" resolved to 162.253.133.38
```

I tried socket and httpget again; symptoms persist. Hmm.

dckc, 2019-11-30:
for context, https://github.com/dckc/dnsanchor is the project where I'd like to use xs for deployment on linux.

dckc, 2019-11-30:
This is what I get when I build the socket example; maybe you could try the `mc.so` in here and see if you get similar symptoms?

[socket-ex.zip](https://github.com/Moddable-OpenSource/moddable/files/3907577/socket-ex.zip)

dckc, 2019-12-02:
I tried turning off ipv6 and that made the symptoms go away.

I hope that diagnosis suggests a quick fix. I'd rather not have to live with that work-around, though I'll understand if you consider this problem out of your scope.

closed 2020-01-24 by mkellner