Moddable-OpenSource/moddable #299: Serial over wireless (WiFi and BLE) -- advice needed

mikew67, 2019-11-30:
For my educational robot project, I need to establish a wireless serial communication channel.  Plan to do one over WiFi first, then BLE.  Consider that I know network protocols, WiFi and BLE.  What I need is advice on how to do this in the Moddable SDK, with an ESP32.

- For WiFI, your Telnet example is instructive.  Poked around the example code, and the module code for CLI, Telnet, etc.  Was able to establish Telnet communication between a Telnet app on my Mac and your "shell" (CLI) on the ESP32, including tracing and debugging the receipt of "commands" on the ESP32.

- This protocol (e.g., Mac sends a "command line" consisting of a command word followed by some params, waits for a response. Target receives same and sends a response line, followed by a prompt character) would be sufficient for my purposes.  Should I just inject my own CLI commands to implement this, or do you recommend another approach.

- In the case of BLE, I didn't see a "Telnet-like" service in your examples.  Any suggestions? (Presume I could define a proprietary characteristic which is a line of text, with the appropriate properties for reading/writing between peripheral and central, with acknowledgment.)

TIA,

Mike

bfriedkin, 2019-12-02:
Mike -

In the case of BLE, take a look at the [ble-friend](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/network/ble/ble-friend) example app. The app was originally written for the [Adafruit BLE Friend](https://www.adafruit.com/product/2479) device, but it is really just a BLE app that implements the central role of the Nordic UART service.

I've attached a simple app that implements the peripheral role.

Please refer to our [BLE reference documentation](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/network/ble/ble.md) for information on available APIs and callbacks.

Regards,
Brian

[ble-friend-server.zip](https://github.com/Moddable-OpenSource/moddable/files/3912111/ble-friend-server.zip)

mikew67, 2019-12-03:
Brian,

Thanks.  Checking it out now.

Mike

mikew67, 2019-12-03:
Brian, _et al,_

ble-friend-server works just fine and meets much of my need for programming the ESP32 as a peripheral (server) endpoint for UART traffic.  Possible improvements I might consider:

- There's probably an official BLE service for bidirectional serial traffic now.  Can switch to that vs. the proprietary UUID in this code.
- Will check to see what BLE version is supported by the ESP32 and Espressif's IDF.  Some of the new stuff in BLE 5, e.g., direction finding, ranging and mesh communication would be really useful for mobile robots.

Thanks for the help,

Mike

mikew67, 2019-12-03:
Hmm... Looks like Espressif has been busy.  Their BLE stack is now "mesh certified". See
[this.](https://www.espressif.com/en/news/ESP_BLE_MESH_SIG_Certified)

mikew67, 2019-12-06:
To add a postscript to this issue, here's a detailed analysis of "Serial over BLE".

[Bluetooth Low Energy Serial: A Valid Design Strategy?](https://punchthrough.com/serial-over-ble/?__s=pkufmirqdfcziiz64utp

Note that it includes a reference to a second article on maximizing serial throughput under various constraints of power, latency, etc.  Hoe folks find this useful.

bfriedkin, 2019-12-07:
Mike -

Thank you for the follow-up. I had actually read part two of the throughput series:

[Maximizing BLE Throughput: Use a Larger ATT MTU](https://punchthrough.com/blog/posts/maximizing-ble-throughput-part-2-use-larger-att-mtu)

You might find that interesting, especially if you require sending data larger than the default 23 byte MTU size from an ESP32 client. If that is the case, you can call the `BLEClient` `exchangeMTU()` function to request a larger size. Typically mobile BLE client apps request a larger MTU.

Regards,
Brian