Moddable-OpenSource/moddable #301: Build failing on Raspberry pi

diogoviannaaraujo, 2019-12-05:
Hi, i’m having problems building moddable sdk on a Raspberry pi 3B+, it gives and undefined reference error to `__atomic_*`

I a amazing luck I read about adding -latomic on make for this kind of problems and I was able to compile most of XS I guess. But when compiling xsbug with mcconfig in the end of tools.mk it still gives those errors.

Is there a way to configure -latomic for mcconfig?

Thanks

diogoviannaaraujo, 2019-12-05:
ok, adding -latomic to LINK_LIBRARIES in tools/mcconfig/make.x-lin.mk got me through this part, compiling takes a while on Raspberry, will return with more news when its finished

diogoviannaaraujo, 2019-12-05:
Ok,
That solved, just to clarify, to build on Raspberry Pi 3B+ i had to add `-latomic` to these files:
- xs/makefiles/lin/xxl.mk
- build/makefiles/lin/tools.mk
- tools/mcconfig/make.x-lib.ml
inside LINK_LIBRARIES

diogoviannaaraujo, 2019-12-08:
Getting it to compile the ESP32 is causing more problems, first get a working toolchain, compilation isn't working  and both versions from espressif I could find for arm are giving segmentation faults. The one that got further is:
https://dl.espressif.com/dl/xtensa-esp32-elf-linux-armel-1.22.0-87-gb57bad3-5.2.0.tar.gz

make: *** [/home/pi/moddable/build/tmp/esp32/debug/helloworld/makefile:418: /home/pi/moddable/build/tmp/esp32/debug/lib/xsCode.c.o] Error 4

This is the error

diogoviannaaraujo, 2020-05-25:
Returning on this, would the -latomic break building in x86? Have only raspberry pi at the moment so i cant confirm.

jins-tkomoda, 2020-11-19:
Sorry, it is written in Japanese, please refer to the following article. I have been able to develop in this procedure.
https://qiita.com/komde/items/10792368ced4e807449c

phoddie, 2020-11-19:
@jins-tkomoda - Thank you for sharing this great article. I browsing it, I found that you are setting the baud rate for the debugger:

```
$ env ESPBAUD=1500000 DEBUGGER_SPEED=1500000 mcconfig -v -d -m -p esp32/m5stick_c
```

A short time ago, we updated the build to set the debugger and upload baud rate automatically for M5Stick C (and others). So maybe you can just write this:

```
mcconfig -v -d -m -p esp32/m5stick_c
```

Here's the change:

https://github.com/Moddable-OpenSource/moddable/blob/78a56c8821c3a12da2979f2f7c97ea8dd19d0d0d/build/devices/esp32/targets/m5stick_c/manifest.json#L2-L5

When you update the Moddable SDK be sure to also rebuild the tools. You can find instructions [here](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Keeping%20Up%20To%20Date.md).

jins-tkomoda, 2020-11-19:
@phoddie Thank you for your information. I'm glad the work around has been essentially solved.

diogoviannaaraujo, 2020-11-19:
Thanks @jins-tkomoda I've been using the same -latomic workaround, wonder if that could be set conditionally in anyway, editing the file makes updating a little hard.

Do this have to with the 32 bits architecture, would a more modern 64bit OS on raspberry pi 4 be able to do this atomic operations without this workaround. Im still pretty new to make and stuff.

phoddie, 2020-11-20:
Good news. We verified that adding `-latomic` is harmless on build targets that don't require it. So, we've applied the change. It should no longer be necessary to apply patches to build on Raspberry Pi.

phoddie, 2020-11-20:
Closing this now that the changes are committed. If you run into related problems, please reopen with the details. Thank you.

closed 2020-11-20 by phoddie