Moddable-OpenSource/moddable #302: Configurable ESP32 base project

PaulFreund, 2019-12-20:
First off, thanks for your great work!

Would it be possible to add a function to mcconfig where I can specify my own ESP-IDF base project to use instead of the xsProj. Should be enough to just make the pasth possible as a command line argument and let it point to the bundled xsProj by default. Also (maybe as an option) to not overwrite the sdkconfig if it already exists (I think that might already be the case). This way I integrate my own ESP-IDF project into the moddable toolchain and functionality without needing to fork and change anything on the base project as far as I can see.

bfriedkin, 2019-12-20:
@PaulFreund -

Thank you for the compliment!

The Moddable build for ESP32 allows developers to specify a `sdkconfig` file in the application manifest. The [modClock](https://github.com/Moddable-OpenSource/moddable/tree/public/contributed/modClock) app uses this feature:

```
"build": {
     "SDKCONFIGPATH": "$(MODDABLE)/contributed/modClock/sdkconfig",
},
```

The `sdkconfig` file provided by the application overrides the base configuration defined in `xsProj/sdkconfig.defaults`. Entries in the application `sdkconfig` file that do not appear in the base configuration are added to the build. When an entry appears in both files, the build uses the entry defined by the application `sdkconfig` file. Perhaps this feature can provide what you need.

The `sdkconfig.defaults` file in `xsProj` is not overwritten by the Moddable build. A copy is made into a temporary build directory.

Regards,
Brian

PaulFreund, 2019-12-20:
Thanks for the explanation about the sdkconfig, that is indeed the behaviour I require. 

In addition to that I need to make my own base ESP-IDF project, most likely by copying the original ESP32 xsProj folder to a path outside of the moddable folder and customizing it. Then I would like to specify this path in the mcconfig call. This way I can integrate my existing C code with the XS engine.

PaulFreund, 2020-01-09:
Slight bump after the holidays ;)

Moddable-OpenSource, 2020-01-09:
Looking at `tools/mcconfig/make.esp32.mk`, try changing the `=` to `?=` on the line:
`PROJ_DIR_TEMPLATE = $(BUILD_DIR)/devices/esp32/xsProj`.

Then you can set an environment variable `PROJ_DIR_TEMPLATE` to point to your esp-idf project.

PaulFreund, 2020-01-09:
Sounds good, I will try it out beginning of next week and report back. If this works as intended can we put it in the mainline? That way I could use moddable as is and update without needing to modify.

mkellner, 2020-01-09:
Yes, we can.

phoddie, 2020-11-17:
This has ben idle for a long time. @PaulFreund, if you have an update please let us know. If not, we'll close this out.

PaulFreund, 2020-11-18:
Hi, very sorry for not follwoing up. I had a shift in priorities and was not able to continue evaluation at this time. I think the proposed solution was working but I can't rememnber for sure. I hope you agree I close this ticket.

closed 2020-11-18 by PaulFreund