Moddable-OpenSource/moddable #303: non-determinism of %GeneratorFunction%[[Prototype]]

dckc, 2019-12-24:
refs:

@agoric/harden assertion
https://github.com/Agoric/harden/blob/0a5185ee224b98e75623a2f88143d118e0849eaa/src/anonIntrinsics.js#L230
and issue https://github.com/Agoric/harden/issues/47

spec fix, which was already fixed
https://github.com/tc39/ecma262/issues/1707

cc @erights

dckc, 2020-01-03:
I'm not able to confirm this fix https://github.com/Moddable-OpenSource/moddable/commit/f7c0a4eb846b7cefdf05bad240b37a002c93f3c8

more in https://github.com/Agoric/harden/issues/47 I hope...

phoddie, 2020-10-14:
This appears to be related to the SES shim. Remember that SES does not want compartments to access Function constructor through function definitions. So the XS linker changes `Function.prototype.constructor` into a function that always throws. 

The report links to [anonIntrinsics.js](https://github.com/Agoric/harden/blob/0a5185ee224b98e75623a2f88143d118e0849eaa/src/anonIntrinsics.js). The `sampleAnonIntrinsics` function seems to check the intrinsics hierarchy before applying the SES constraints. 

If you execute that function in `xst` or `xsnap`, there are no problems. If you execute that function in the Moddable SDK runtime (with `"strip": []`), it fails because the SES constraints have already been applied.

closed 2020-10-14 by phoddie