Moddable-OpenSource/moddable #307: Error while compiling REPL example: mxClosures undeclared

bartmichu, 2019-12-29:
Hello.

I'm using Ubuntu 19.10 with TTGO T8 ESP32-WROVER board. Environment and toolchain configured as per Getting Started guide, with different project paths. It works for basic examples like ```pins/simpleAnalog``` or ```pins/blink```.

Now I would like to interactively play around with my board so I'm trying to compile ```js/repl``` example:

```shell
cd $MODDABLE/examples/
ESPBAUD=460800 mcconfig -m -p esp32
...
...
...
# cc repl.c.o
/home/xxx/Documents/projekty/moddable/moddable/examples/js/repl/esp/repl.c: In function 'xs_repl_eval':
/home/xxx/Documents/projekty/moddable/moddable/examples/js/repl/esp/repl.c:63:99: error: 'mxClosures' undeclared (first use in this function)
  fxRunScript(the, fxParseScript(the, &aStream, fxStringGetter, mxProgramFlag), &mxGlobal, C_NULL, mxClosures.value.reference, C_NULL, C_NULL);
                                                                                                   ^
/home/xxx/Documents/projekty/moddable/moddable/examples/js/repl/esp/repl.c:63:99: note: each undeclared identifier is reported only once for each function it appears in
make: *** [/home/xxx/Documents/projekty/moddable/moddable/build/tmp/esp32/release/repl/makefile:531: /home/xxx/Documents/projekty/moddable/moddable/build/tmp/esp32/release/repl/repl.c.o] Error 1
```

Any suggestions, please?

bartmichu, 2020-01-25:
The same problem with ```esp``` platform, I tried on Moddable One and NodeMCU.

mvincentmiller, 2020-01-26:
I got through everything up to this point, I'll look into mxClosures...

mvincentmiller, 2020-01-26:
Appears to be a case of deprecation:

https://github.com/eco747/foton/search?q=mxClosures&unscoped_q=mxClosures

mvincentmiller, 2020-01-27:
I got this to run with an older version, if anyone is interested in exploring it please message me. It looks like they deprecated it because it wasn't quite stable enough for an example, but it's a super cool entry point into the platform.

bartmichu, 2020-01-27:
@mvincentmiller I'm very interested, can you share some more details about how to get it working?

phoddie, 2020-05-13:
The `js/repl`example is building on top-of-tree. Confirmed with macOS, ESP32, and ESP8266 built targets. Closing this issue. If you run into a problem building or running, please open a new issue. Thank you.

closed 2020-05-13 by phoddie