Moddable-OpenSource/moddable #309: Moddable 2 Screen Not Coming On

sblindt, 2020-01-05:
Hello! I recently purchased a Moddable 2 from your website.  I can flash all of the example projects to it, scan wifi, see "Hello World" in XSBUG.  Everything seems to work great until I try the Piu Balls example.  The program gets flashed and I can see the device being used in XSBUG but the screen never comes on showing any interaction. I've followed the setup guide for Moddable 2 including trying to manually set the back light but nothing seems to get the screen to respond. Do you have any advice for getting the screen working?

andycarle, 2020-01-06:
Greetings!

I'm sorry to hear that you're having issues with the screen on your Moddable Two. Let's see if we can get you back up and running quickly.

The best guess from the team here is that you're building from outdated sources. There have been some recent changes to the tree to support the newer version of the Moddable Two, and building without those changes could explain the symptoms you're seeing.

To that end, could you please verify that you've pulled the latest sources from github and then try doing a clean build of the Moddable SDK? You'll need to go into the `$MODDABLE/build` directory and delete `bin` and `tmp` entirely. Then rebuild the Moddable SDK tools for your platform following the instructions in the [Getting Started](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md) document. Finally, rebuild your example app and deploy it to the Moddable Two device.

If that does not work, we'll investigate other potential causes. While we test every device just before it goes out the door, there's always the possibility that something was damaged in shipping. If that turns out to be the issue, we can get you a new device shipped out quickly.

Cheers!
&nbsp;- Andy

sblindt, 2020-01-07:
Thank you for the quick response, Andy! That was exactly it. Pulled in the changes from GitHub and rebuilt the tools and it works like a charm. Thanks again!

closed 2020-01-07 by sblindt