Moddable-OpenSource/moddable #311: Crypto library

diogoviannaaraujo, 2020-01-11:
Hi, im trying to port a JWT sign function to create a password for google not core, but im having some problems find the correspondent code for Node's:

`crypto.createSign(method).update(input).sign(key, 'base64')`.

I've been reading Moddable Crypt module but the RS256 Class I've found is really cryptic..
Is there some module that could solve this or should I make a C code wrapper?

Thanks

diogoviannaaraujo, 2020-01-11:
Im pretty close, read a little more undiscovered what PKCS1_5 was about, now im just having some stack overflow problems when using its sign() function.

diogoviannaaraujo, 2020-01-11:
Ok, I was able to generate the JWT to log into google cloud iot, but the process is taking 10 seconds! Need to find out how to improve that performance.

wilberforce, 2020-01-20:
> I've been reading Moddable Crypt module but the RS256 Class I've found is really cryptic..

The esp hardware has acceleration for this sort of thing so if you use the esp-idf lib it will use the hardware acceleration. Much faster than doing in C!

bartmichu, 2020-09-20:
@diogoviannaaraujo would you share some code? I'm planning to play around with GCP in near future.