Moddable-OpenSource/moddable #316: XS misinterprets `await` in arrow functions

jugglinmike, 2020-01-23:
In synchronous arrow functions defined at the "top level" of module code, XS incorrectly allows AwaitExpressions:

    (() => { await 0; });

When such functions are invoked at the top level or within a synchronous function, they cause XS to crash. When they are invoked from within an async function, the functions synchronously return the first awaited value (without executing further statements).

The following code demonstrates the behavior when evaluated with [the `xst` binary](https://github.com/Moddable-OpenSource/moddable-xst) as provided by [jsvu](https://github.com/GoogleChromeLabs/jsvu):

    var evaluated = false;
    var a = () => {
      await Promise.resolve(0);
      evaluated = true;
    };

    (async () => {
      print('a(): ' + a());
      print('evaluated: ' + evaluated);
    })();

Output:

    a(): [object Promise]
    evaluated: false

phoddie, 2020-05-13:
Confirmed that XS 10 also accepts this syntax.

phoddie, 2020-08-26:
XS now rejects this:

```
SyntaxError: missing ;
```

closed 2020-10-08 by phoddie