Moddable-OpenSource/moddable #317: WSL Instructions for Windows

bterlson, 2020-01-23:
Windows comes with Linux these days, and so MINGW is less than ideal. Would it be possible to update the [setup instructions](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md#windows) for ESP32 to use WSL rather than MINGW? [This doc](https://techcommunity.microsoft.com/t5/internet-of-things/easily-connect-espressif-devices-to-azure-iot/ba-p/1121262) might help!

andycarle, 2020-02-20:
@bterlson Thank you for the suggestion! I've put together a version of our build that uses WSL in place of MINGW. It is not quite ready for prime time, but it would be very helpful to have more people give the instructions a go and see if the build works on their systems.

If you want to try it, you need to replace $MODDABLE/tools/ mcmanifest.js and $MODDABLE/tools/mcconfig/nmake.esp32.mk with the versions from the zip attached below. Then follow the instructions in the included markdown file, which would replace the [Windows ESP32](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md#esp32-windows) section of the Getting Started guide.

Please let me know if you have a chance to try this out.
&nbsp;- Andy

[esp32-wsl.zip](https://github.com/Moddable-OpenSource/moddable/files/4228357/esp32-wsl.zip)

wilberforce, 2020-02-21:
@andycarle - Thanks for this.

for my WSL I seem to be set up as root:
```
root@office-pc:~# pwd
/root
root@office-pc:~# whoami
root
root@office-pc:~# ls esp32
esp-idf  xtensa-esp32-elf  xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz
```

in the Developer Command Prompt for VS 2017 I have:

```
C:\Users\rhys\source\repos\www-m5-mash>set WSL_IDF_PATH=/root/esp32/esp-idf

C:\Users\rhys\source\repos\www-m5-mash>set WSL_ESP32_TOOLCHAIN=/root/esp32/xtensa-esp32-elf

C:\Users\rhys\source\repos\www-m5-mash>mcconfig -d -m -p esp32
C:\Users\rhys\Projects\moddable\build\tmp\esp32\debug\xsProj
        1 file(s) copied.
# cc xsHost.o (strings in flash)
The system cannot find the path specified.
NMAKE : fatal error U1077: 'C:\Users\rhys\esp32\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc' : return code '0x1'
Stop.
```

This seems to be coming from:
`TOOLS_ROOT = $(BASE_DIR)\esp32\xtensa-esp32-elf`

Is there a tweak to mcconfig (where is it?) that will give more verbose output for nmake?

andycarle, 2020-02-21:
@wilberforce Thanks for trying this out--that's very helpful! 

You can `set VERBOSE=1` in Developer Command Prompt to get a bit more output from the build. That has generally been sufficient for me while I've been debugging this.

That particular error sounds like you did not get the Windows toolchain installed or you have it in a non-standard location. After step 4 in the esp32-wsl.md document, you should have a Windows toolchain at `C:\Users\rhys\esp32\xtensa-esp32-elf\bin\`. If your ESP32 Windows toolchain is elsewhere, you can point to it with the `TOOLS_ROOT` environment variable. 

If you have the MINGW tools bundle (from our public Windows build instructions), you will still need to do step 4 and download the stand-alone version of the toolchain.

wilberforce, 2020-02-21:
@andycarle 
Thanks - sorry I had missed step 4 as I thought I only needed the WSL linux too chain.

I've got further now - that exec is getting found, however now have:

```
C:\Users\rhys\source\repos\www-m5-mash>mcconfig -d -m -p esp32
C:\Users\rhys\Projects\moddable\build\tmp\esp32\debug\xsProj
        1 file(s) copied.
# cc xsHost.o (strings in flash)
The system cannot find the path specified.
NMAKE : fatal error U1077: 'C:\Users\rhys\esp32\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc' : return code '0x1'
Stop.

C:\Users\rhys\source\repos\www-m5-mash>mcconfig -d -m -p esp32
C:\Users\rhys\Projects\moddable\build\tmp\esp32\debug\xsProj
        1 file(s) copied.
# cc xsHost.o (strings in flash)
In file included from C:\Users\rhys\Projects\moddable\xs\sources/xsCommon.h:41,
                 from C:\Users\rhys\Projects\moddable\xs\sources/xsAll.h:41,
                 from C:\Users\rhys\Projects\moddable\xs\platforms\esp\xsHost.c:38:
C:\Users\rhys\Projects\moddable\xs\platforms\esp/xsPlatform.h:102:10: fatal error: lwip/netdb.h: No such file or directory
 #include <lwip/netdb.h>
          ^~~~~~~~~~~~~~
compilation terminated.
NMAKE : fatal error U1077: 'C:\Users\rhys\esp32\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.EXE' : return code '0x1'
Stop.
```

Looks like I night need to upgrade the SDK or clean down all the temp files?

andycarle, 2020-02-21:
@wilberforce Yeah, you'll definitely want to do a clean build of everything (including the Moddable SDK tools). Just delete your `%MODDABLE%\build\bin` and `%MODDABLE%\build\tmp` directories. Then run `%MODDABLE%\build\makefiles\win\build.bat` to rebuild tools. 

I'll add a note in the next version of the instructions to clarify that a clean build is required when switching between these build methods.

wilberforce, 2020-02-21:
@andycarle 
Do you think you could detect the presence of the WSL_IDF_PATH var and siliently support both MING and WSL with same nmake script?

andycarle, 2020-02-21:
@wilberforce Yes, definitely. My goal for the moment is just to make sure this method works on a few different systems (and to get feedback on if it is an improvement worth making). Then we'll decide what to do with this. :)

wilberforce, 2020-02-21:
@andycarle 

After clean down - still gettting:

```
C:\Users\rhys\Projects\moddable\xs\platforms\esp/xsPlatform.h:102:10: fatal error: lwip/netdb.h: No such file or directory
 #include <lwip/netdb.h>
```

Looks like and SDF version issue? https://github.com/Moddable-OpenSource/moddable/issues/128

andycarle, 2020-02-21:
@wilberforce Agreed, probably an ESP-IDF version issue. 

The WSL build assumes ESP-IDF v3.2.2, which you should end up with after going through the instructions (`git clone -b v3.2.2 --config core.autocrlf=false --recursive https://github.com/espressif/esp-idf.git`). Top-of-tree Moddable SDK esp32 builds in generally most likely won't work correctly with any ESP-IDF version older than the 3.2 line.

wilberforce, 2020-02-22:
Thanks. I have managed to get a build. I'm using the M5Stick-C which is good as a I found an issue with the baudrate - it needs 1500000.

I needed to change - nmake.esp32.mk:313

`ESPBAUD=$(UPLOAD_SPEED) `

```
%windir%\sysnative\wsl.exe touch ./main/main.c; ESPBAUD=$(UPLOAD_SPEED) DEBUGGER_SPEED=$(DEBUGGER_SPEED) MODDABLE=$(WSL_MODDABLE) PATH=$$PATH:$(WSL_ESP32_TOOLCHAIN) IDF_PATH=$(WSL_IDF_PATH) BATCH_BUILD=1 DEBUG=1 IDF_BUILD_DIR=$(WSL_IDF_BUILD_DIR) SDKCONFIG_DEFAULTS=$(WSL_SDKCONFIG_DEFAULTS) make flash
```

I also have an possibly unrelated issue - the display is no longer working.. that might be a change to the idf version from my previous build...

wilberforce, 2020-02-22:
With regards step 13  - instead  having two copies of the idf.... one in windows the other in WSL.

instead of:

set WSL_IDF_PATH=/home/<username>/esp32/esp-idf

We can use of the fact that WSL can access the windows file system so this is possible:
set WSL_IDF_PATH=/mnt/c/Users/rhys/esp32/esp-idf

Or - is this what your default set up is effectively doing? Would this be slower?

andycarle, 2020-02-24:
@wilberforce That's great that you got a build going! Agreed that there are some issues (with this build method or the original) on the M5Stick-C's upload speed. I have not seen any issues with the screen on Moddable Two, but I'll test an M5Stick-C at the office tomorrow.

On your last comment, yes, that is exactly what the default setup is doing. It is accessing your Windows clone of the ESP-IDF through the /mnt/c/ path. But because it is crossing over the VM boundary to do that, it is a bit slow. (Going the other way--cloning only in Linux and doing the Windows parts of the build by accessing the Linux files--is even slower. I tested.). It is quite a bit faster to clone twice, as described in step 13.

barbiani, 2020-02-28:
> Windows comes with Linux these days, and so MINGW is less than ideal. Would it be possible to update the [setup [...]

I actually ran WSL as the real linux and went to the linux section of the readme.

Xsbug needs a X server running.

andycarle, 2020-02-28:
@barbiani Right, this is the other good option here: just live entirely in Linux within WSL and follow all of the Linux instructions. 

I've had good luck running that way in conjunction with a Windows build of xsbug, to get around needing an X server. If there was serious interest in that workflow, we might be able to distribute a standalone binary of xsbug for Windows so that one doesn't have to get the Moddable SDK build going on Windows just to build xsbug.

barbiani, 2020-02-28:
Ever thought of an adapter for vscode?

andycarle, 2020-04-16:
We now support native ESP32 builds on Windows using CMake and Espressif's idf.py utility. Please see [this blog post](https://blog.moddable.com/blog/moddable-sdk-improvements-for-esp32-projects/) for details.

This update removes the requirement for either MinGW or WSL in our Windows ESP32 builds, so I'm closing out this issue.

@barbiani : I do have an interest in additional support for vscode, but no concrete plans. There's a little bit of discussion in [this issue thread](https://github.com/Moddable-OpenSource/moddable/issues/257) as well.

closed 2020-04-16 by andycarle