Moddable-OpenSource/moddable #329: pubnub example broken

bartmichu, 2020-02-01:
Pubnub example compiles and uploads fine but then fails with:

```
Wi-Fi connected to "redacted"
IP address 10.0.0.17
Generating time-based UUID.
Subscribing...
/home/redacted/redacted/moddable/modules/network/socket/lwip/modSocket.c (705) # Break: (host): unsupported type for write!
```
I'm using esp8266 board.

lprader, 2020-02-03:
Thanks for the report. I've committed a fix that will be included in our next open source update. 

If you want to try it out before then, replace `$MODDABLE/modules/network/services/pubnub/pubnub.js` with [this](https://github.com/Moddable-OpenSource/moddable/files/4150542/pubnub.txt). Note that you'll have to change the file extension to .js (because GitHub apparently doesn't allow attaching .js files here).

bartmichu, 2020-02-04:
That was quick, thank you - I can confirm that it works fine for me with your fix.

andycarle, 2020-02-25:
This fix is now live in the open source repo.

closed 2020-02-25 by andycarle