Moddable-OpenSource/moddable #330: Misinterprets arrow functions without parentheses in many expressions

mzgoddard, 2020-02-07:
XS misinterprets some expressions with arrow functions as valid syntax but instead should report a SyntaxError or early ReferenceError.

The following expressions are not valid JavaScript. Adding parentheses around the arrows would make all of these expressions valid. The values XS currently reports (as returns or thrown exceptions) would match if they had parentheses.

```js
() => {}.call // function call (){[native code]}
() => {}['call'] // function call (){[native code]}
() => {}`` // undefined
new () => {} // TypeError: ?: new 9: no constructor
() => {}() // undefined
() => {}(Date) // undefined
() => {}(Date.now) // undefined
+() => {} // NaN
!() => {} // false
delete () => {} // true
void () => {} // undefined
typeof () => {} // function
2 << () => {} // 2
() => {} instanceof Function // true
() => {} || 1 // function  (){[native code]}
'abc' in () => {} // false
() => {} ? true : false // true
```

I tested this running the xst binary installed by jsvu.

```sh
$ ~/.jsvu/xs -v
XS 9.0.3
$ uname -a
Darwin beep3.localdomain 18.7.0 Darwin Kernel Version 18.7.0: Thu Jan 23 06:52:12 PST 2020; root:xnu-4903.278.25~1/RELEASE_X86_64 x86_64
```

phoddie, 2020-05-13:
Confirmed these are active with XS10.

phoddie, 2020-08-04:
These all generate an error now. (Thank you for the excellent find!)

closed 2020-08-04 by phoddie